opentx/opentx #6504: Option to bind without telemetry on X9D+ with 2.2.3 might not work.

Docteh, 2019-06-15:
**Describe the bug**
The telemetry options when binding don't seem to work on an X9D+

**To Reproduce**
Steps to reproduce the behavior:
1. Bind and choose `Ch1-8 Telem OFF`
2. Power up XSR holding its button
3. Press ENT on taranis to exit binding mode.
4. Power up XSR without touching the button
5. Look at telemetry tab.

**Expected behavior**
I'd like to bind without telemetry. Its currently interfering with a video transmitter. 
Rebinding without telemetry works fine with a TaranisQ X7 running OpenTX 2.2.3, but that transmitter is not mine.

**General information**
 - OpenTX version is 2.2.3, tried 2.2.4N703 as well
 - Issue is with a Frsky X9D+, 
 - XSR is running Fport firmware.

3djc, 2019-06-15:
I tried with regular firmware 2018-11-06, it works as expected (this latest release specifiocaly adds that)
Flashed F.Port version (lastest I found was 2018-04-27) : works as expected too

Tested with x7 2.2.3, as I do not have access to x9d+, but hardware is extremely similar

Docteh, 2019-06-15:
I flashed XJT_NONEU_170317.frk to the internal XJT module. Appears to work on the bench. No test flight yet.

It might be related to that my X9D+ is manufactured back in 2015 April

closed 2019-06-15 by 3djc