opentx/opentx #6509: Lua scripts, menu button on Taranis X7 does not work running OpenTX 2.3

vkolotov, 2019-06-17:
I know that 2.3 is not release yet, but this might help you anyway.

I'm testing OpenTX 2.3 and Betaflight lua scripts (latest release - 1.3.0). I can switch to a display with lua script and I can see my PIDs, however I cannot switch to other screens or save changes. Effectively "menu" button does not work, when I push it (either long or short) nothing happens.

* Taranis X7
* OpenTX 2.3 (8c56a131ede013104bc89f6cf2a82063faeb5061)
CMAKE OPTIONS: -DDISABLE_COMPANION=YES -DLUA_COMPILER=NO -DLUA=YES -DSUPPORT_D16_EU_ONLY=NO -DTRANSLATIONS=EN -DTARANIS_INTERNAL_PPM=NO -DGVARS=YES -DSHUTDOWN_CONFIRMATION=NO -DMULTIMODULE=NO -DRAS=YES -DOVERRIDE_CHANNEL_FUNCTION=YES -DPCB=X7 -DPPM_UNIT=PERCENT_PREC1 -DHELI=NO
* Betaflight lua scripts (1.3.0)

vkolotov, 2019-06-17:
Could it be something to do with this?
https://github.com/opentx/opentx/pull/6502

3djc, 2019-06-17:
Could you test again after pulling ?

vkolotov, 2019-06-17:
Yep, it is working now.

3djc, 2019-06-17:
Thx for reporting and testing !

closed 2019-06-17 by 3djc