opentx/opentx #6513: OpenTX v2.2.3 on Horus X12S with R9M module and Emergency Mode

PeterBeretta, 2019-06-17:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
My Horus X12S running OTX 2.2.3 goes into Emergency Mode when the R9M module power setting is set to "Auto <=1W". When in Emergency Mode, the RX remains bound but there is no control. This cause my plan to crash.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on the radio with the R9M module set to "Auto <=1W"
2. Wait about 1-2min without switching on the RX.
3. The radio goes into Emergency Mode

**Expected behavior**
Emergency Mode is not supposed to render the Horus X12S inoperable i.e. if bound to an RX and Emergency Mode occurs, servos do not respond to inputs.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version 2.2.3
 - Hardware Horus X12S
 - OpenTX Companion 2.2
- R9M Firmware FW_Ranger9_TX_FLEX_20190117

3djc, 2019-06-17:
As explained already, you do not have enough power to sustain 1w, emergency mode triggers because of that, but cannot solve that issue. There is nothing that can be done software wise to solve this, you need to ensure there is adequate power to sustain 1w

Emergency has been successfully tested with R9M and control remains

PeterBeretta, 2019-06-17:
There could be an issue with my Horus hardware then because when my Horus goes into Emergency Mode, there is no control, despite the fact that RX is still bound i.e. LED is green on the RX.

3djc, 2019-06-17:
Yes, there is clearly an issue with your hardware setup as a whole, you should not emergency each time power ramps up !

closed 2019-07-14 by 3djc