opentx/opentx #6518: External 5v Power Enable

robthomson, 2019-06-20:
It would be useful to have a facility on a 'per model basis' to enable power output on the SPORT connector on the rear of the radios like the X10S.

The rational is as follows:

**Wireless Trainer Port.**

It is possible to add a micro ppm receiver internally within the radio.  This receiver could be easily soldered to the SPORT pins on the X10/X10S for power; and additionally to the PPM - IN on the trainer jack that is right next to it.

You could even achieve this with a 3d printed 'clip on' device on the rear of the case.

The problem is that the SPORT connector does not deliver any power unless:

SPORT_UPDATE_ON()   

is called.

So.. a feature within opentx on a per model bases to enable the ports power supply would be usefull.  This would in effect turn on / off a wireless trainer function.  

End result.. super easy to achieve wireless head trackers; or wireless trainer using third party RF systems other than bluetooth!

I do not believe this function will be too difficult to add - and am personally in the process of attempting to get a working tool chain on my pc running so that I can compile. (somewhat painfull to achieve)

If per chance anyone fancies making the change in the mean time; would be a great feature to add to 2.3 milestone!

kilrah, 2019-06-20:
Ï think I already have an issue open for this but was planning to do it system wide, not per model.

robthomson, 2019-06-21:
That would also work.  Main reason I was thinking 'per model' is to enable easy on/off state - essentially saving battery power when rx not in use!

Net result is you can solder in a permanent rx inside the radio - but still retain trainer port function as rx work if 'powered up'

robthomson, 2019-07-01:
Well.. be damned if my tool chain will work.

Going to have to wait for someone to magic this feature in - either that or rebuild my pc!!

OpenUAS, 2019-08-21:
Nice idea, and per model even better! Our toolchain does work, so does the PC... however our time-machine is broken at the moment... @kilrah if you already have a branch with something, a link here and it is easier for someone (with spare time left ;) ) to extend than to start from scratch...

robustini, 2019-09-06:
I simply enable the external module to have the active power on the appropriate connector, in the end it is the same thing.
I put the receiver inside the module housing, with a female servo socket glued to the upper edge of the housing, so that by removing the male plug I can still install an external module.

WillyZehnder, 2019-09-26:
I have a external module installed and use the power of that connector for the internal Trainer-receiver - and it works. But that receiver is always running as soon as I activate the external module - also when I don't need a Trainer-signal. 
So the idea of robthomson to activate the SPORT-connector per model is much smarter, because the receiver could just be switched on if it's needed.

bsongis, 2020-03-24:
Done in 2.3.6

closed 2020-03-24 by bsongis