opentx/opentx #6522: Startup switch position

jj-uk, 2019-06-22:
I need to change a single switch startup position to be DOWN.

At the moment, it only seems possible to change to switch up, or not checked, by using the radio.

Maybe it would be better to highlight the switch, flash the cursor, then allow the user to change the switch, auto-detecting which switch and which position it is in. The same say as switches are selected throughout all other radio setup screens.

3djc, 2019-06-22:
Lets see : no radio details mentionned, no version mentionned either ...

jj-uk, 2019-06-22:
Whooooops, sorry, Updated :)

3djc, 2019-06-22:
Then you set all your switches like you want them en push ENT long

jj-uk, 2019-06-22:
Thanks, yes, but I don't want to check all switches, only 3 switches. The way this works is inconsistent with how all other switch selection is done in the radio menus.

3djc, 2019-06-22:
Then you enter the line and activate (or not) switches individually. Most user check all their switches pos, and that can be set in one (long) click. Your proposal when you need to set many switches (X9E has 16 for example) would just be a very painfull experience

closed 2019-06-22 by 3djc

jj-uk, 2019-06-22:
The current action of holding ENT to set all switches to current state should not change.

What about adding:
Hold ENT when an individual switch is highlighted. The physical switch is then read, and the position is saved as the position to check at startup.

This is the same as the current action of "all at once", but for individual switches.

kilrah, 2019-06-22:
No change is needed. When you highlight a single switch you can select whether thatswitch is checked or not at startup already. It doesn't matter that all switches are stored in one go.