opentx/opentx #6523: Lua scripts, up/down buttons on X LIte Pro not work running N30 FW

meccup, 2019-06-22:
Hey @3djc, I'm testing now X Lite Pro. Looks like there might be a bug similar to this one: https://github.com/opentx/opentx/issues/6509

Effectively I cannot move cursor. I can switch pages with left/right stick movements, but up/down stick input does not work, cursor stays on the same field.

3djc, 2019-06-22:
What script are you testing with ? Up/down is used in the FrSky config script, could you try one of those ?

meccup, 2019-06-22:
That was Betaflight scripts.

I've tried wizard script, looks like up button is working, but down is not.

3djc, 2019-06-22:
Found the issue for down, will push a fix. The BF scripts might need to be updated however

closed 2019-06-22 by 3djc

3djc, 2019-06-22:
What action are up/down supposed to perform in BF scripts ?

meccup, 2019-06-23:
Yep, I can confirm N31 fixes the issue. I can now move cursor in the BF lua script. Thank you.

Up/Down keys are moving cursor from one field (betaflight config, PID, VTX etc) to another.

One thing I cannot figure out how to do in X Lite Pro is display switching. I set up two display for a model, one for BF lua, another one for telemetry. I can launch BF lua by long press down (little stick), but then whatever I do, it does not switch to my second display which is telemetry display.

meccup, 2019-06-23:
Haha, just found this: https://www.youtube.com/watch?v=lXCcg2S3_RE

You need to hold right button and use joystick to switch telemetry screens/displays.

3djc, 2019-06-23:
shit + long left/right

commit 74086231 (2019-06-22): Fix #6523 (at least the OTX part)