opentx/opentx #6526: Multiprotocol FlySky

BergshamraFraggel, 2019-06-27:
OpenTX 2.2.3 Taranis X9D
FlySky 9X9 protocols are 8 channel protocols but OpenTX seems to send 16 channels and the parameter cannot be changed.
The signal "spill over" between the channels of the receiver, while the control channel 5 reacts Channel 4 in certain situations.
It looks like all protocols except DSM are locked to 16 channels, even the FrSky D16 8ch.
[image]
[image]
[image]
[image]

kilrah, 2019-06-27:
AFAIK the Multiprotocol module is supposed to do the job of sending what's appropriate for each protocol, like it does remapping and special functions.

BergshamraFraggel, 2019-06-27:
Okey....
Why does the same module work with that receiver if I turn the module knob to position 1 and set the radio to PPM ch 1 - 8 ??

kilrah, 2019-06-27:
Module probably does things differently in that case. 

On some protocols the "option" can tell the module how many channels to send, you could try playing with that in "Custom" mode to see if it's the case there.

BergshamraFraggel, 2019-06-27:
If I had the opportunity to change the settings to ch 1-8 in the same way as for DSM, it would probably solve the problem too!
But it is perhaps easier for you to explain how I set the parameters to use customs settings ??
What are the values of FlySky 9X9 ??

kilrah, 2019-06-27:
No, most likely not.

Values for Custom are in the multiprotocol documentation.

closed 2019-07-03 by BergshamraFraggel