opentx/opentx #6527: Adding Points to a Curve

aceplayer55, 2019-06-27:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
When you add points to a curve, OpenTX moves the existing points.  I find that inconvenient.

**Describe the solution you'd like**
A simple improvement would be to add all new points to the curve at x=0 and at its respective y value.  You could then place the new point(s) as desired.
I would expect this to be simple to implement.
Thank You.

kilrah, 2019-06-27:
It moves the points in X obviously since by default interfal is fixed. What you describe would mess up the existing curve whenever adding / removing points, while the current interpolates as required precisely so the curve doesn't change.

aceplayer55, 2019-06-27:
You are right that it would be better to maintain spacing when there is no curve, initially or if the curve is a straight line.
What I am suggesting is that when a non-linear curve has points added to it, the user might prefer to keep the location of the current points.  So in this case, the program could add (or offer the option of adding) all the new points at x=0 and the current value of y.  This maintains both the current curve and spacing and inserts new points at x=0 as requested.

kilrah, 2019-06-27:
That is not trivial, currently:
- By default points have hardcoded X, you need to explicitly switch to custom to enable user-defined X
- Points have to be stored in order
- Points cannot overlap

If anything had to be redone, I'd rather see a more common "insert before/insert after" that puts a new point at an X in the middle of 2 existing ones.

elecpower, 2019-06-27:
Provided the curve is user-defined X, Companion will permit the insertion of points via ctl+click on the curve line without moving existing.

aceplayer55, 2019-06-27:
Kilrah said "If anything had to be redone, I'd rather see a more common "insert before/insert after" that puts a new point at an X in the middle of 2 existing ones."
Yes - That would be what I am requesting.  I am just suggesting putting new points at x-0 and letting the user move them.

Elecpower = So then it is desirable.  Does having it in Companion preclude it from being in OpenTX?

Thanks (both of you)

kilrah, 2019-06-27:
It already is in companion. 

Lot more work to do on the radio, but possible of course.

aceplayer55, 2019-06-27:
Inserting new points at x=0 eliminates the need to have a ctrl key !!

kilrah, 2019-06-27:
Nothing to do with the control key, but with doing all the coding works 3 (soon 4) times for all the different UIs for all the platforms we support. Companion does it once for all. 
Putting new points at 0 is actually significantly more complicated for nothing.

closed 2020-03-23 by stale[bot]