opentx/opentx #6528: no undervoltage Warung at LUA Script?

Basti77, 2019-06-28:
Hello, yesterday my Xlite went out in flight. (battery empty)

Luck in sight. I could land even more via the telemetry connection of the notebook.  (Andruplane)

The way it looks, you don't get any acoustic warnings from the device when a Lua script is running, which is itself replay sounds. 

TX: X-Lite (first model)
OTX 2.23

LUA script: Yapuu Telemetry 
https://github.com/yaapu/FrskyTelemetryScript 

I have to check my batteries to see if they are defective because of the deep discharge. 

otherwise it is difficult to adjust the case without opening the transmitter to connect an adjustable power supply.

3djc, 2019-06-28:
What alarm did you put ? There is no reason for that to happen if your alarm is set right

kilrah, 2019-06-28:
Lua or not doesn't affect battery monitoring at all. Make sure your low battery alarm is set correctly and that the voice file is present. 

The only reason that could be thought of is if the script puts out SO many plays that it completely saturates the voice queue... but then I'd think it's more likely you'd miss the warning even if it's played becasue you can't concentrate on listening to so many things.

You can always add additional alarms using logical switches and special functions if you want.

closed 2019-06-28 by kilrah