opentx/opentx #6529: opentx compilation error with GCC 8.3.0

lachesis, 2019-06-28:
**Describe the bug**
I cannot compile opentx default branch (2.3) on Arch Linux using GCC 8.3.0.

**To Reproduce**
```
git clone https://github.com/opentx/opentx
cd opentx
cmake .
make
```

**Expected behavior**
No build errors.

**Actual behavior**
```
[ 21%] Building CXX object radio/src/CMakeFiles/firmware.dir/gui/212x64/model_display.cpp.o
/home/user/build/opentx/radio/src/gui/212x64/model_display.cpp: In function 'void menuModelDisplay(event_t)':
/home/user/build/opentx/radio/src/gui/212x64/model_display.cpp:242:87: error: cannot bind packed field 'g_model.ModelData::screens[((int)screenIndex)].TelemetryScreenData::lines[((int)lineIndex)].FrSkyLineData::sources[c]' to 'source_t&' {aka 'short unsigned int&'}
  242 |             source_t & value = g_model.screens[screenIndex].lines[lineIndex].sources[c];
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
make[2]: *** [radio/src/CMakeFiles/firmware.dir/build.make:1454: radio/src/CMakeFiles/firmware.dir/gui/212x64/model_display.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:150: radio/src/CMakeFiles/firmware.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

**General information**
 - OpenTX version: git hash 4412d812e3b9a25875d48e1361fc96f9ea457283

I hit a similar problem (with a different error message) when trying to compile opentx-companion. The error message is:
```
/home/user/build/opentx-companion/src/opentx-2.2.3/radio/src/gui/480x272/model_gvars.cpp: In function 'void editGVarValue(coord_t, coord_t, event_t, uint8_t, uint8_t, LcdFlags)':
/home/user/build/opentx-companion/src/opentx-2.2.3/radio/src/gui/480x272/model_gvars.cpp:26:30: error: cannot bind packed field 'fm->FlightModeData::gvars[((int)gvar)]' to 'gvar_t&' {aka 'short int&'}
   26 |   gvar_t & v = fm->gvars[gvar];
      |                ~~~~~~~~~~~~~~^
In file included from /home/user/build/opentx-companion/src/opentx-2.2.3/radio/src/gui/480x272/layout.h:25,
                 from /home/user/build/opentx-companion/src/opentx-2.2.3/radio/src/datastructs.h:778,
                 from /home/user/build/opentx-companion/src/opentx-2.2.3/radio/src/myeeprom.h:24,
                 from /home/user/build/opentx-companion/src/opentx-2.2.3/radio/src/opentx.h:305,
                 from /home/user/build/opentx-companion/src/opentx-2.2.3/radio/src/gui/480x272/model_gvars.cpp:21:
```

People are talking about this on the ArchLinux User Repository package discussion page for opentx-companion: https://aur.archlinux.org/packages/opentx-companion/?comments=all&O=0&PP=10

kilrah, 2019-06-28:
As explained in the build instructions only 4.7.2 is supported for building firmware.

bsongis, 2019-06-28:
I compile the simu with gcc 8.3 here (Manjaro which is Arch based)
But for the firmware we still use an old compiler.

bsongis, 2019-07-08:
I updated Manjaro today, now I have gcc version 9.1.0 and I encounter the same error

commit ab8e6026 (2019-07-08): Fixes #6529
closed 2019-07-08 by bsongis