opentx/opentx #6533: 2.3.0N35 Switches SC and up not working on XLITE Pro in companion

lshems, 2019-06-29:
Create a new model, select any place where you can choose a physical switch as source (I tested on flightmodes and inputs). SA and SB get 'autoselected' on moving. Switch SC and up not.

If selecting manually or by companion, the switch doesn't have any effect on the function it is used on on the radio (through the sim).

lshems, 2019-06-29:
Checked hardware definition, is ok. 

Don't know where to find the raw input ADC values, page seem to be gone, so can't check there.

lshems, 2019-06-30:
I can't confirm radio side issues but they could well be there. If so I think it is important to label radio as well on this bug.

If it is on the radio as well, this is going to crash planes, as people are using the nightlies as there is nothing else.

3djc, 2019-06-30:
If I added companion, it is because I checked on the radio

pmackenzie, 2019-06-30:
Might be related, but I found that SD↑ does not work in logical switches in Companion, but !SD↑ does. See attached image.  It does work correctly on the radio (X9 lite)
SD down position is processed correctly. SE has the same error.

pmackenzie, 2019-06-30:
Couldn't upload the file, here is a link to the image on RCG

https://www.rcgroups.com/forums/thumbgallery.php?do=showattach&u=14199

commit 29acf1b8 (2019-07-01): Fixes #6533
closed 2019-07-01 by bsongis