opentx/opentx #6535: GVAR won't Increment/Decrement in steps smaller than MIN value

xoedusk, 2019-06-30:
**Describe the bug**
The Special Function "Adjust GVx > Increment" does not allow me to input an INCREMENT value that is lower than the manually-set minimum value of the Global Variable. Example, if I set up a Global Variable with a minimum of 40 and maximum of 150, the lowest increment value I can enter is 40. I want small steps, even negative steps, but I cannot do it.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a Global Variable with min 40 and max 150
2. Create a Special Function using any switch, select Adjust GV# (newly created G.V.), then select Increment.
3. Try to enter an increment value less than 40, such as 2. It will revert back to the minimum value of 40.

**Expected behavior**
I should be able to enter any increment number I want, even negative, so long as its absolute value doesn't exceed some reasonable "range"

**Screenshots**

Defining the GVAR:

[image]

Selecting Increment:

[image]

About to click out of the textbox to set an increment of 30:

[image]

But it reverts back to 40:

[image]

**General information**
 - OpenTX version 2.2.3
 - Hardware X10S
 - OpenTX Companion 2.2.3 on MacOS

elecpower, 2019-07-01:
Fixed in 2.2.4 and 2.3

closed 2019-07-01 by elecpower

3djc, 2019-07-01:
👍