opentx/opentx #6547: Broken link in manual/documentation for 2.2

haumovie, 2019-07-05:
On the page https://opentx.gitbooks.io/manual-for-opentx-2-2/content/companion/flashing.html
the link to "Sdcard" (https://opentx.gitbooks.io/manual-for-opentx-2-2/content/companion/sdcard.md) is broken. 

On the page https://opentx.gitbooks.io/manual-for-opentx-2-2/content/advanced/audio.html
the link to "Companion SDcard"(https://opentx.gitbooks.io/manual-for-opentx-2-2/content/advanced/sdcard.md) is also broken.

Possibly, both should link to https://opentx.gitbooks.io/manual-for-opentx-2-2/content/sdcard.html

3djc, 2019-07-14:
This manual could realy use a steward ! If someone is interested in maintaining it, please contact us

closed 2019-07-14 by 3djc

haumovie, 2019-07-14:
I guess so. As a rare breed of once-were-tester, I actually enjoy writing documentation and will Co sider being a steward of sorts. My primary shortfall is, that I have only been flying RC fixed wing for 1 year, only just 7 days ago got my first radio with opentx and so by no means a subject matter expert. Thinking commences.....