opentx/opentx #6548: X9Lite - 2.3.0N37 - Delay between RSSI warnings and telemetry lost

ijustwannafly, 2019-07-05:
**Describe the bug**
X9Lite with OTX 2.3.0N37.  XSR-M bound in ACCST D16.  There is a significant delay from the time "RSSI Low/RSSI Critical" plays to the time "Telemetry Lost" is played.  

**To Reproduce**
Steps to reproduce the behavior:
1. Power up X9Lite and select model bound to XSR-M.
2. Power up XSR-M and verify connection is established ie trigger beeper on quad
3. Disconnect power from XSR-M.
4. X9Lite first plays "RSSI Low/RSSI Critical" but RSSI bars on main screen are gone indicating no RX connection.
5. After RX connection is lost "Telemetry Lost" is played signficantly delayed from actual action.

**Expected behavior**
Based on X-Lite on 2.2.3 - "Telemetry Lost" should be played immediately if RX connection is lost.

**General information**
 - OpenTX version 2.3.0N37
 - Hardware X9Lite
 - OpenTX Companion 2.3

kilrah, 2019-07-06:
Noticed this with an X7 yesterday...

ijustwannafly, 2019-07-06:
@kilrah 

It picks up quickly as it should on the "Telemetry Recovered" action but definitely leaves the RSSI callouts almost too dangerous to trust for now.

I also flashed N40 last night and it still persists but I expected that as N40 was already available before I posted.

raphaelcoeffic, 2019-07-31:
Here is the deal:
- the RSSI value used for warnings, checks for working telemetry, etc, is actually an average of the last 4 values received.
- in case the connection to the receiver is lost, depending on the number of "0db" packets passed by the module, you might get different results:
    - avg of `[80, 80, 0, 0]` is `160/4 = 40`.
    - only if the average is `0`, you will get "telemetry lost".
- after a while, the value is "expired" (1s after the last RSSI packet has been received), but the "telemetry lost" will only come later when `telemetryWakeup()` is called.

In 2.2.4, the code was:
```
      if (data == 0)
        telemetryData.rssi.reset();
      else
        telemetryData.rssi.set(data);
```

basically effecting the "telemetry lost" immediately.

ijustwannafly, 2019-07-31:
@raphaelcoeffic It's not so much that the "telemetry lost" is playing delayed.  It's that the "telemetry lost" playback/notification (as in past versions) has almost consistently happened when the rx connection is lost.

To add some other information to this, on my x9lite on N52 as of today, the delay is still present but ONLY with ACCST receivers.  My r9m lite and r9m lite pro on ACCESS firmware and bound to an ACCESS receiver will go straight to "telemetry lost" upon rx power loss/connection loss.

raphaelcoeffic, 2019-07-31:
> the delay is still present but ONLY with ACCST receivers
Well, that probably means that the ISRM behaves differently in ACCST and ACCESS mode, which is not really good.
Moreover, the issue has been seen on ACCST only transmitters with 2.3 (self compiled, no nightlies).

raphaelcoeffic, 2019-07-31:
Anyway, this works for me:
```
@@ -171,11 +171,17 @@ void sportProcessTelemetryPacketWithoutCrc(uint8_t origin, const uint8_t * packe
      if (g_model.rssiSource) {
        TelemetrySensor * sensor = &g_model.telemetrySensors[g_model.rssiSource - 1];
        if (sensor->isSameInstance(TELEM_PROTO_FRSKY_SPORT, instance)) {
          telemetryData.rssi.set(data);
          if (0 == data)
            telemetryData.rssi.reset();
          else
            telemetryData.rssi.set(data);
        }
      }
      else {
        telemetryData.rssi.set(data);
        if (0 == data)
          telemetryData.rssi.reset();
        else
          telemetryData.rssi.set(data);
      }
    }
    else if (dataId == R9_PWR_ID) {
```

commit 5b516b82 (2019-08-02): Fix #6548
closed 2019-08-02 by 3djc