opentx/opentx #6551: Taranis 9X Lite Wheel issue

PonyOny, 2019-07-07:
On occasions when you turn on the Taranis 9x Lite the wheel button will not work when rolling, it will only click, you have to restart the radio several times until the wheel works correctly and allows you to navigate up & down through menus.

 - OpenTX version: 2.3.0N40
 - Hardware: Taranis 9X Lite
 - OpenTX Companion: 2.3.0N39

P-I-Engineer, 2019-07-07:
same for me

bsongis, 2019-07-08:
I am searching for a possible cause, I never encountered it on my end

bsongis, 2019-07-08:
I found something funny, if you turn the wheel during the startup sequence, the radio will shutdown instead of start

P-I-Engineer, 2019-07-09:
N41 seems to be working

Harpalyke, 2019-07-09:
I also have this issue with my x9lite. Started to notice it when I flashed N33 a couple of weeks ago but I didn't get the chance to update the firmware since.

I've tried the N41 build and it still seems to be a problem on my radio. It's not happening as frequently as N33 though, I was seeing it happen about 8 out of 10 times before. 

One thing that I have noticed is that when it boots and the scroll wheel isn't working if I leave the radio idle for a minute or so the scroll wheel then works again.

I had thought it might have possibly been a hardware issue at first but the scroll wheel always seems to work in the bootloader menu so I'm guessing it probably is an issue on the software side.

bsongis, 2019-07-11:
Is it possible one of you contact me by email. I think the debounce delay value I chose is not optimal on some radios, I would like to check which is the value which would work for you.

bsongis, 2019-07-11:
Unless the

closed 2019-07-11 by bsongis

P-I-Engineer, 2019-07-11:
will do tonight, thanks

Harpalyke, 2019-07-12:
Seems to have done the trick. I rebooted the radio about 10 times in a row and the scroll wheel is now responsive 100% of the time after it boots. Thanks.

glideasw28, 2019-07-12:
I have the exact same issue as "Harpalyke" above. Not using nightly though. 
Running official firmware update from frsky (2019-06-26) 2.3.0
Scroll works one in ten starts. Scroll will work if left idle for a minute.

Thanks.

bsongis, 2019-07-12:
@Harpalyke would you confirm that the issue was fixed between N41 and N42?

P-I-Engineer, 2019-07-12:
N44 is working

Harpalyke, 2019-07-12:
@bsongis, yep, seems fine on my radio now also.

commit a8373c17 (2019-07-11): Attempt to fix #6551