opentx/opentx #6552: DSM2 binding broken in 2.1.9

numanair, 2019-07-08:
**Describe the bug**
Same as https://github.com/opentx/opentx/issues/3628
I am creating a new issue because the last release for avr radios like the 9X still has this issue.

**To Reproduce**

**Expected behavior**

**General information**
 - OpenTX version 2.1.9
 - Turnigy 9X
 - OpenTX Companion 2.1.9

Can this same fix https://github.com/opentx/opentx/commit/d97e41335a239dcaa25036f3a20db8da4613aa82 be applied to this older version? I assume implementing this fix for my own use requires setting up the whole build environment.
Older versions (2.0.17) work on my radio just fine.

kilrah, 2019-07-08:
This fix only affects the sky9x upgrade board.

2.1 basically brought nothing more than 2.0 for AVR, so maybe you can just stay on 2.0...

numanair, 2019-07-08:
Okay, I'll just use 2.0 as long as firmware is still provided. Hopefully that does not change.

kilrah, 2019-07-08:
Setting milestone just in case it ever happened...

numanair, 2019-07-09:
It appears this is actually broken in 2.0.19 as well. The light on my module does not come on with this firmware.

kilrah, 2019-07-10:
What module is this exactly? The ones affected by the issue do not have an LED in the first place.

numanair, 2019-07-12:
This is a lp4 module, DSMX IIRC. It was pulled from one of those basic RTF radios. It's been a long time since I touched this radio so I don't remember the specifics. I can open it up if there's a way to get more details from the board.

closed 2019-10-11 by bsongis