opentx/opentx #6553: Nightly Warning Removal (opion)

robthomson, 2019-07-08:
When using 'nightly' version of the firmware... on startup you get a prompt to 'warn' you that the firmware is not safe.

I totally get the reason why this is the case - but it can also be rather annoying in certain scenario.

e.g.  I would like to use the 2.2 branch with vario support via csrf

This means I have to use a nightly.

It would be good to have an option to turn this off at own risk.

Either via a 'companion' check box.. or an option on the radio - maybe buried away in the system menu somewhere.

kilrah, 2019-07-08:
We have discussed it and decided not to implement, the warning is there for a reason, and skipping it is not the end of the world..

closed 2019-08-14 by raphaelcoeffic

shamufpv, 2019-09-16:
The latest nightly did not give me a warning screen. Everything seems to be working though. Should I be concerned?

3djc, 2019-09-16:
You probably have an RC that does not trugger the warniing. please check your exact version in radio/version screen