opentx/opentx #6555: Assign extra PCB inputs available already in X9-Lite to SG and LS and RS.

lshems, 2019-07-08:
The X9-Lite has 5 switches and a pot. On, the board, provisions have been made for easy diy extension with 2 pots (or switches) and one switch (to switch a 2 pos or toggle).

I would suggest to include these already in the available hardware options on the X9-Lite.

The pots could go on the standard places that are also used on a taranis X9D.
The switch could be anywhere, but I guess lots of people would make it a pushbutton for launch (DLG) or similar on the left side.

By doing so, this would prevent a renumbering of the source list afterwards, which will prevent a lot of rework at that time.

Also, currently, the entire switch 'stuff' in Opentx seems a little tricky, as lot's of bugs have been reported using them.

So, perhaps better take the dive now, and include already ??

Suggestion only, do as you wish.

kilrah, 2019-07-08:
> On, the board, provisions have been made for easy diy extension with 2 pots (or switches) and one switch (to switch a 2 pos or toggle).

> By doing so, this would prevent a renumbering of the source list afterwards, which will prevent a lot of rework at that time.

Once again this was an afterthought from FrSky that was not in the first board version we got, and we were not notified of this addition. 

So source list renumbering is now necessary, so we can't do it anymore. The radio has been out there for a couple of months and that would break everyone's models.

A complete rework of how switches work is planned so this recurring trend isn't an issue anymore.

> Also, currently, the entire switch 'stuff' in Opentx seems a little tricky, as lot's of bugs have been reported using them.

Care to elaborate?

lshems, 2019-07-08:
Elaborate: #6533 , and my incorrect one I just closed. 
Got your hint: lot's is an overstatement; Sorry for that.

Got a little carried away as the current way of using the sources in lists without functions to find the ones you need for special functions in LUA for instance is really a nightmare, trying to make multiradio scripts.

I do however think the argument of the radio being out there a tricky argument. There is no official OpenTX out yet that supports the X9-Lite.
I do understand it though. 

Will these extra inputs be added nonetheless, in 2.3, or will we have to wait for 2.4??

kilrah, 2019-07-08:
The firmware that's on the radios out of the box is OpenTX. We intend to not break the compatibility with that, so no more format changes.

That will be for 2.4. With the new system. 

To be fair, with what FrSky pulled off with the new radios, the fact that they did not want to supply samples for the X9-Lite to any member of the team other than their employee, and even worse that they did not allow them to see the technical docs before public release, it serves them right if some stuff is missing and we are not going to make any particular effort to expedite it.

lshems, 2019-07-08:
Get your point, I can read between the lines. Thanks.

P.S. which models did get switch updates? 
P.P.S. I won't close as 2.4 is still some point in future, not like never.

3djc, 2019-07-08:
X7, x10 and x12 get extra switches, the horus getting also two more pots

MikeBland, 2019-08-09:
I was originally told by FrSky that the X9Lite was only going to ship with erskyTx. Quite late on they told me it was going to be available with a choice of openTx or erskyTx. So, while I requested the spare pins be brought out to pads, I didn't know I needed to pass this on to openTx.
In erskyTx, I have a dynamic method of handling the switches. Each switch has a unique hardware value, that remains fixed across firmware releases (and radios). I create a switch mapping table, and an unmapping table. This table is updated if you set an extra switch is available in the hardware menu.

Mike

3djc, 2019-08-10:
Mike, it was obviously FrSky job to communicate the info, not yours, but likely because of their evil plan to drop out open source, they haven't.

We are adding in 2.4 mechanisms that should enable us to deal with things like this much faster (ie not relying on a major version change)

lshems, 2019-11-02:
No longer on 2.4 milestone??

thkn-hofa, 2019-11-24:
> No longer on 2.4 milestone??

I'm also interrested

alexdibi, 2020-04-18:
I have an x9 lite bit i don't see any extra PCB input likes qx7

closed 2021-01-14 by stale[bot]