opentx/opentx #6557: Add Mixer value widget to TX

xoedusk, 2019-07-09:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**

There is already a convenient widget for viewing Output values on my Horus. But there is not similar widget for mixer values. I care about seeing the “logical” values, not what my FBL wants to see. 

**Describe the solution you'd like**

I would like a widget just like the Output one, except with the Mixer outgoing values. 

**Describe alternatives you've considered**

Could already be done in Companion, but I don’t use Conpanion as much as just straight up raw TX configuration. 

**Additional context**

I’m using Horus X10S running OpenTX 2.2.3

kilrah, 2019-07-09:
Is the servo/mixer monitor not enough?

[image]

xoedusk, 2019-07-09:
I’m my opinion, no. If that were sufficient then there’d be no reason to have an Output widget. But there is an output widget. Okay, that logic doesn’t quite hold up. But hopefully it’s clear what I mean.... why have an output widget but no mixer widget?

By having that as a widget one could view Timer, RPM, etc along with the Mixer values from a single page. That is the functionality that is missing 

If that were sufficient here’s be no reason to add

3djc, 2019-07-09:
Mixer widget is certainly not mainstream, and can be done out of LUA using the exemple provided

closed 2020-04-04 by stale[bot]