opentx/opentx #6558: Play Track list: Long press and inertia

xoedusk, 2019-07-09:
**Is your feature request related to a problem? Please describe.**

I downloaded and am using the Amber sound pack. SO MANY TRACKS. Even with the spin wheel on my X10S, it takes forever to get to the center of the alphabet. Oh sweet Jesus. 

**Describe the solution you'd like**

Two solutions. Both can be enacted. 

1. Enable long press to bring up a simple pop up  list of single numbers and letters. (# A B C, one per line). Selecting that number or letter or character jumps you to the corresponding part in the track listing. Much faster. I’d be happier. 

2. Enable inertia scrolling in the track list. Is that what it’s called? In some menus I can spin faster to super-fast jump ahead. But with the track list  ow it’s purely one line per click. Make it like changing the output min/max values on the Output page. That’s fast. 

**Describe alternatives you've considered**

Hmm....

**Additional context**

I’m on X10S with OpenTX 2.2.3.

xoedusk, 2020-01-05:
I’m still interested in this feature being added.

kniuk, 2020-04-08:
I think adding inertia scrolling would be great. Maybe a playback of a file when you pause for 2 seconds ?

xoedusk, 2020-05-21:
Yes, playback after 2 secs would be great.

nmaggioni, 2020-11-19:
Ping!

prodzprodz, 2021-01-13:
Is this hard to implement?

Realy, it would be god to have fast scroling when scroling trough sonuds. Something similar when scroll trough letters when write model name, if you spin faster, you jump ahead.
It realy takes forever to get to the center of the alphabet when change sonds on TX.

nmaggioni, 2021-07-13:
Pong.