opentx/opentx #6564: Saving models and settings to Radio can delete all models if Hardware Patch is enabled

jims1234, 2019-07-12:
I could not find this bug listed nor could I find a Companion specific issues area so I apologize if it's been reported somewhere else as a Companion issue here on GitHub. The problem is serious in that it will consistently delete all models and settings in the radio whenever the option to  "Patch HW settings from profile"  is enabled just before a "Write Models and Settings to Radio" operation is requested .. This Write step will consistently delete all models and settings in the Radio IF another  Settings option is also enabled . To set this option read in a .otx model file and in the upper right select "Edit Radio Settings and at the bottom right os that page Enable the "  in teh Radio Setup 
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

jims1234, 2019-07-12:
Please help me delete the original issue above and use this text below:

I could not find this bug listed nor could I find a Companion specific issues area so I apologize if it's been reported somewhere else before as a Companion issue on GitHub. This is my 1st post. The problem is serious in that it will consistently delete all models and settings in I believe any Taranis radio whenever the option to "Patch HW settings from profile" is pre-enabled just before a "Write Models and Settings to Radio" operation is requested.. 
This write step will consistently delete all models and settings in the Radio IF another Settings option is also pre-enabled.  To set this option read in any .otx model file and in the upper right of the dialogue box listing the models select "Edit Radio Settings”  and then at the bottom right of that page Enable the “ Store calib. and selected settings in Profile” option.. the combination of both options set will cause the model delete issue on both companion 2.2.3 and Companion 2.2.4..  

I’m using the Windows version of Companion on Win 10 64 bit with a Taranis Q 7XS and as a new user found it very frustrating in that it was initially intermittent since I did not recognize the correlation with the two options as indicated  above.. Somehow this issue got “entered” here after the 1st sentence and way Before I had a chance to complete it.. for ref with screen grabs..  It’s also posted on RCGroups.com under a thread with this title: “Anyone else had a problem saving .otx files with OpenTx and Companion 2.2.3? to a Tar(anis) by jims123

kilrah, 2019-07-12:
Click the 3 dots on top right of the post and Edit.

jims1234, 2019-07-12:
Ok thanks kilrah.. been there done that ..   v/r jims

PR #6567 (2019-07-13): Fix HW patch feature
HW patch was not working at all, this fixes the basic functionnaly and also fixes #6564
commit f8e7b368 (2019-07-13): Attempt to fix #6564
closed 2019-07-13 by 3djc