opentx/opentx #6565: sportTelemetryPop() unpredictable result when value is > 67108860

dgatf, 2019-07-12:
In sportTelemetryPop() packets with values > 67108860 produces unpredictable results. The highest byte of the packet is unusable

bsongis, 2019-07-15:
ah oui t'as raison, bizarre, j'espère qu'on n'a pas quelque chose sur 24 bits

kilrah, 2019-07-15:
More data needed. What are you using as source? 
There are scripts like ardupilot data displays that transfer non-numeric data over all bytes and work just fine...

raphaelcoeffic, 2019-08-24:
@dgatf we need more input, and something to reproduce, as it works for others.

dgatf, 2020-02-28:
Sorry for taking a while to answer

I've tested on opentx 2.3 an I can't reproduce it. It must has been an error on my side. There is no issue for any value

FYI is for this project [https://github.com/dgatf/msrc](url)

Thanks

closed 2020-02-28 by dgatf