opentx/opentx #6566: LUA for X10S don't recognize a long press on PgUp/Dn key.

m-rik-prog, 2019-07-13:
The classic code provided as an example in some of the scripts of the various SD Cards:

if event == nil then
    return 2
elseif event == EVT_PAGE_BREAK or event == EVT_PAGEDN_FIRST then
    selectPage(1)
elseif event == EVT_PAGE_LONG or event == EVT_PAGEUP_FIRST then
    killEvents(event);
    selectPage(-1)
end

Works correctly with a short press or a long press on the single PAGE key of the X7.
It also works correctly on X12S, (but it has two separate keys for PgUp and PgDw).
It work in only one direction on the single PgUp/Dn key of the X10S even with the latest OpenTX 2.2.4 version.
Not even by modifying the script and trying other combinations/condictions.
On X10S which is currently the most popular radio, with a short press or a long press nothing changes and you can only go ahead in the menus of the script.

3djc, 2019-07-13:
Version ?

m-rik-prog, 2019-07-13:
As explained "even with the latest OpenTX 2.2.4" (but obviously also with precedent versions).

3djc, 2019-07-13:
This is fixed in 2.3 (and scrips should ideally be adapted to the new virtual events for easier cross radio operations)

closed 2019-07-13 by 3djc

m-rik-prog, 2019-07-13:
Thank you.
To all my users who complain I will say that it will be solved in version 2.3 of OpenTX.
But they will surely ask me "when" it will be released to version 2.3
Is there an indicative forecast?

3djc, 2019-07-13:
Unfortunately, we do not do eta, especially when some of the elements are totally out of our control

kilrah, 2019-07-13:
You can use other keys e.g. SYS and TELE as left/right.

  elseif (event == EVT_TELEM_FIRST) 
    selectPage(1)
  elseif (event == EVT_SYS_FIRST)
    selectPage(-1)

m-rik-prog, 2019-07-14:
Thank you Kilrah for the idea of this workaround.
Added and tested, work (obviously also on X12S).
Updated pdf instructions of the application.

m-rik-prog, 2019-07-14:
Note for developers: the long press problem exist also running the Windows Firmware Simulator.

kilrah, 2019-07-14:
simulator runs the same code, so of course if the required event isn't defined in firmware it won't work on both.