opentx/opentx #6568: Less ram available for LUA scripts on X9D than for X7

m-rik-prog, 2019-07-14:
I have an application that I've managed to reduce to a minimum and compress so that the same precompiled application (.luac) can work not only on the Horus X12 & X10 but also on the Taranis X7 (even more than once since after the first run it was no longer working the second time because it obviously leaves some areas of copper occupied and does not free them all and I had to compress further giving up some functions).

However, on the Taranis X9D it doesn't run even on the first run and the message of insufficient memory appears. Note: it runs perfectly with the emulator that uses the ram of the PC.

Is there a possibility to free up some RAM by modifying the system configuration in order to deactivate some function that occupies more RAM on the Taranis X9D than on the Taranis X7?
(unfortunately more than that I can't compress the application that has 14 pages to modify more than 300 parameters).

3djc, 2019-07-14:
x7 has more ram because the screen resolution is smaller, and yes, ram is an issue on taranis,  there is some gain that can be made using luac, but overall, everything that can be done has been done already

closed 2019-07-14 by 3djc

m-rik-prog, 2019-07-14:
Sidenote: X7 also lacks some code. For example, when an insufficient memory error is generated by running a LUA script, the message "Not enough memory" that appears on the X9D does not appear on the X7 screen, which simply does not run the lua/luac code. But I don't think this affects the RAM space but only the ROM space.

bsongis, 2019-07-14:
> because it obviously leaves some areas of RAM occupied and does not free them all

are you sure, isn't it only RAM fragmentation?

lshems, 2019-07-15:
@m-rik-prog

Check my blog on rcgroups. It might give some ideas