opentx/opentx #6576: External module powered while loading on X-Lite Pro

Tim2309, 2019-07-17:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

kilrah, 2019-07-17:
That is not OpenTX-related, this is due to a(nother) issue with the "PMU" FrSky has added to these radios.

closed 2019-07-17 by kilrah

Talha909, 2019-07-24:
I have the same issue and write down the email to frysky.  They sent the update file of PMU.  I have installed the update and after that this issue resolve.

Tim2309, 2019-07-24:
Last week they send me this in an email:
„There is an issue of the radios powering modules when the USB or charge circuit is live. I do not believe there is a fix just yet. Please remove modules when charging the radio. I know it is inconvenient at the moment, and hopefully will be addressed soon.“

Can you share the file with us?

Talha909, 2019-07-24:
Sure. I will share the file. I have the same issue . But after update this issue resolve. Please give me your email address. I will send you the file.

Tim2309, 2019-07-24:
Please send it to tim(at)phaea.de
Thank you

Talha909, 2019-07-24:
I have attached the file over here. Please check.
https://www.rcgroups.com/forums/showthread.php?3080337-FrSky-New-Taranis-X-Lite-S-and-Pro-X9-Lite-R9M-Lite-and-Pro-MPM-Lite/page194
[autoupdate.zip](https://github.com/opentx/opentx/files/3426716/autoupdate.zip)

Tim2309, 2019-07-24:
Thank you for sharing, now it works :)