opentx/opentx #6579: SDcard content alignement over radios and versions

lshems, 2019-07-18:
Currently there are different content packs for different radios. Also, the packs are different per version.

I think this is very complicated for a lot of users, especially for those that have multiple radios from different types.

Since the SDCards nowadays are very big, why not make a structure that shares what is the same for all radios in one set of common folders, and stores in radio specific folders what is radio specific. This would allow for ONE generic SDcard content pack for all radios, with a non relevant size penalty.

For the versioning, I suggest to place the SDcard content in a folder per version. If the current OpenTX version of the radio isn't finding a matching folder for the SDcard content, it will look for the content of the previous version, and so on.
This would allow to reflash the radio to a previous version and still being sure all works. With proper radio backups on the SDcard, in combination with a firmware stored on the radio, that assures that you can easily revert to a previous version, even without companion, in the field. 

The Firmwares folder would need to be in the root, and version indifferent.

Just a suggestion, happy to work something out, but first would like to get some responses to the idea.

Shoot!

mpjf01, 2019-07-19:
I'm not qualified to look and check so I'm guessing but I assume that the code is currently written such that it looks in the same place on the SD card for the same information regardless of the radio type. This would allow one version of the code to operate for all the radio types in respect to most of the SD card accesses. Having a different SD card structure for each radio type could mean substantial code changes - depending on how it is structured.

From an end user perspective it would make getting the SD content required a bit easier. I regularly use and update firmware and SD content for a X9, a X9+. a X12S, a X10S and a X-Lite and occasionally bother with my old 9X and 9XR. I don't have any hassle managing this but from the forums I read of some who do.  These are frequently people who have not read (or perhaps) understood the instructional material that is available and I sometimes wonder if there is anything than one could do to assist them apart from automating the entire procedure.

Whilst SD cards these days are capable of holding the data required, consideration needs to be given to those who do not have access to high speed internet where downloading high volumes of data is an issue of both time and cost. Any volume increase could be compensated for by allowing user selection of sound files required rather than  forcing downloads of all languages.

lshems, 2019-07-19:
Opentx is always 'aware' of what radio it is running on, so looking in a radio specific folder would be no issue.

With the versioning it becomes more problematic, as you are linked from companion to a list of packages to download, no info on which version for which firmware version.

It would already help if they would simply synchronise the package versioning with the opentx versioning. Yes, that means identical packages with different names, but who cares?

Just trying to simplify things that go wrong with many users. Lots of them out there ignoring a wrong version alert. They take it as 'normal'.

lshems, 2020-02-14:
Not so stale, as every day it becomes more complex.

closed 2020-11-11 by stale[bot]