opentx/opentx #6580: Betaflight LUA - no telemetry

Tim2309, 2019-07-19:
I use the X-Lite Pro with the full crossfire module, telemetry works well, but the BF LUA doesn't show values and says no telemetry.

With my X10S on opened 2.2 all was fine and the LUA worked with the same quad.

3djc, 2019-07-19:
Well, we have reports suggesting BF scripts do run fine  with 2.3N and Xlite Pro, so that looks linked to XFire. We have asked for XFire gear, but we don't have that yet, so it may be some time before we can effectively help

Tim2309, 2019-07-19:
ok no problem, btw the crossfire lua work without problems

3djc, 2019-07-19:
Even stranger !

Tim2309, 2019-07-19:
anything I can do for testing?

kilrah, 2019-07-19:
Is your XF up to date? Believe they released an update recently.

Tim2309, 2019-07-19:
It’s on 3.00

dmanfly, 2019-07-29:
Hello , having the same issue with X9 Lite  (I guess the X-Lite pro , in the same development stage), on latest nightly today, N55
tested with crsf and with D16, the crsf sends telemetry but the BF lua script is not working
the D16(XSR) , not telemetry, and no lua script, verified that the BF lua and telemetry is working on this quad with X9D 2.2.3

is the telemetry and lua working on other radios on nightly ?
is there any way I can help? test something?

3djc, 2019-07-29:
Not sure I understand. The crsf part I understand.

The D16 as me confused, Are you saying that in D16 mode, your X9 Lite has no telemetry at all and that the LUA scripts do not work ?

dmanfly, 2019-07-29:
yes, on X9 Lite with D16 mode there is no sensors on telemetry screen, and the BF LUA script is not working

Hawxxer, 2019-08-06:
Same Problem X-Lite Pro + 2.3.0N58. What is xfire? The Kiss Lua Script is working fine, but Betaflight lua script is not.
Sidenote:
Updating the crossfire transmitter is not working in 2.3.0N55 dont know about the newest nightly. When I press binding, the Script shows if the Transmitter should be updated, but when pressing Enter nothing happens.

3djc, 2019-08-26:
Issue replicated here, will  be working on it this week

PR #6675 (2019-08-28): 2.3 Fix BF scripts
This fix #6580
closed 2019-08-28 by 3djc