opentx/opentx #6581: X9Lite otx 2.30N49  Modell generating with wizzard  Error at Input Name Ias IGas

hrenz, 2019-07-20:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

hrenz, 2019-07-20:
Testing X9Lite with different Languages German, France, Englich, Czech,.....
otx 2.30 N49
Modell generating with wizzard

English is ok, Input Name is correct
but at France, an German the Input Name is one Character too left 
so its  Eas  insteed of EGas    or    Ias  IGas

[image]

kilrah, 2019-07-20:
SD card up to date with 2.3 contents?

hrenz, 2019-07-20:
ist an V0019 Version, original Tx from Frsky /Engel
All test were made with same SD-card, so its the same wizzard for
en, ge, fr, cz

other users have same issues with german,  Input  one character too left

[image]

3djc, 2019-07-21:
I cannot seem to reproduce
[image]

3djc, 2019-07-21:
This is linked to the wizard process, models created with Otx default are fine

hrenz, 2019-07-21:
Ah, ok  at the German vesion 
And there is an error with spezialchar  äöü too  
allways H8h insteed of Höh  for  Elevator

kilrah, 2019-07-21:
Companion does not support reading/writing special chars to the radio. If you don't want that you can change the wizard translations to set "Hoh" instead of "Höh".

hrenz, 2019-07-21:
ok, where is the wizard translation

kilrah, 2019-07-22:
In the companion translations.

One question, you did not say if you did those in the companion wizard or with the radio wizard, which on is it?.

hrenz, 2019-07-22:
Hallo Andre,
X9Lite otx 2.30N49 Modell generating with wizzard Error at Input Name Ias IGas (#6581)

About the 2 pictures you see at #6581,I did it all with otx germanand  wizzard direct, not from companion 

----------------------------------------------------------------------------I have Companion not tested in this detailbecause in Companion you see Inputname: HohInputsource: Höh
at Mixersource: Hoh, 
But the Simulator in companion is allways in english, and I have not transferd from companion to tx
I think with the cmake file, we can set language-paramter  to de and the companion Simulator will come in german too.This will help all other users very much for SimulatorThis Funktion is missing since yearsBecause Horus X10, X12 has no screenshot funktion, and the screenshot Picture in companion Simulator is english thats not good for writing manulas in cz,  pl,  rus
mfgH. Renz

Verschickt: Mo, 22. Jul. 2019 13:24

bsongis, 2019-07-22:
This issue is fixed now, it will be OK in next nightly build

hrenz, 2019-08-27:
Hallo Bertrand,
openTx v2.3.N68  X9-Lite (small display)

sorry, problem is not solved, look at screenshots
Wrong positinons of the name IGas IH8h, ISei, IQue
 and H8h insteed of Höh  (Elevator)

 mfgH.Renz

Verschickt: Mo, 22. Jul. 2019 21:09

bsongis, 2019-08-28:
I just tested, I see no issue anymore, unless I didn't understand what you did.

[image]
[image]
[image]
[image]

commit 50d00c29 (2019-07-22): Fixes #6581
closed 2019-08-28 by bsongis