opentx/opentx #6582: Shift+Trim not working on Xlite-Pro

MikeNH, 2019-07-21:
**Describe the bug**
Trim for Rudder (Yaw) and Throttle not working Companion and Xlite-Pro
**To Reproduce**
Steps to reproduce the behavior:
1. In Companion, create a new model.
2. Accept all Multicopter defaults (AETR)
3. Click "Simulate Model"
4. Demonstrate that Trim works for Ch 1 (Aileron). Now hold the Shift key (letter "S"). Nothing moves.

**Expected behavior**
The expected behavior when holding the shift key is to see the Rudder Trim (CH 4) move. 
Excerpt from user manual:
_"**Shift Button**
Shift button pressed with Tirm buttons together,Trim for Ail &Ele will shift to Tirm for Rud&Thr."_

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version companion-windows-2.3.0N49
 - Hardware XLite Pro
 - OpenTX Companion Windows-2.3.0N49

kilrah, 2019-07-21:
So that is only in companion or also the radio? Not clear

3djc, 2019-07-21:
Checked, only companion, broken during a refactoring in June

MikeNH, 2019-10-04:
This still is not working in Companion 23.0. When I look at Fix 41682a2 it appears to indicate that this is fixed in release 23.0 but I don't think that is the case. This works as expected on the radio but not Companion.

kilrah, 2019-10-05:
> 2- Demonstrate the problem by pressing the "S" shift key and while holding the S key down press a trim. Nothing happens.

Working here.

commit 41682a2a (2019-07-22): Fixes #6582