opentx/opentx #6584: LUA access to flightmode settings

lshems, 2019-07-21:
We can currently get info about the current or any other flightmode by the getFlightMode() function.

However, there is no access to change for instance the flightmode switch settings on a model, or the flightmode trim settings.

This would be very useful if one would want to create more elaborate and complex wizards for model creation.

A model.setFlightMode and a model.getFlightMode function would be needed.

I am willing to help on it, but need a start to get going on, in the form of a rudimentary function set with some info on how the data is stored.

any place I would be able to find that out?

rdeanchurch, 2019-07-24:
Lua also is missing several things in model.get/setInput().
These include 
curvetype
curvevalue
flightmodes
stickside

All but stickSide are available in model.get/set Mix()

I too would be willing to try to help, but need a starting point.

lshems, 2020-01-21:
Seems not so stale

ridgerunner, 2020-04-04:
I started writing a collective pitch (CP) Heli wizard a few years back but was stopped with the lack of Lua access to creating and editing CURVEs, The API was enhanced to allow creation and editing CURVES so the only missing ingredient at this point is the lack of an API hook for the creation and editing of FLIGHT MODEs.

 My base CP heli model template has evolved significantly over the past seven years and definitely takes advantage of FLIGHT MODEs. IMO a useful heli wizard would require this feature.

PR #7878 (2020-08-28): LUA improvement
Flight mode handling
Inputs input name handling

This closes #6584
closed 2020-08-28 by bsongis