opentx/opentx #6585: X-Lite Pro with OTX 2.3.0N50. S1 and S2 Pot no response.

LOVEChen, 2019-07-22:
Describe the bug
X-Lite Pro with OTX 2.3.0N50. S1 and S2 Pot no response.

To Reproduce
Steps to reproduce the behavior:

Power up X-Lite Pro.
Move S1 and S2 pot watching status bar same time.

General information

OpenTX version 2.3.0N50
Hardware X-Lite Pro
OpenTX Companion 2.3

bsongis, 2019-07-22:
Strange I just flashed it, and it works. Next nightly will come in 2 hours, please confirm if there is still an issue

kilrah, 2019-07-22:
Make sure they are not set to "None" in the Radio setup hardware page, and have been calibrated properly.

drewzh, 2019-07-29:
Fresh install of OpenTx on my X-Lite Pro also gives me non working S1 and S2. The only hardware config option that results in movement is 'multipos'. Is this correct or should they be 'pot'?

3djc, 2019-07-29:
Set then as pot and redo calibration (they need to be moved during calibration too)

drewzh, 2019-07-29:
Thanks @3djc, that solved my issues. First time flashing my X-Lite 👍

LOVEChen, 2019-07-30:
Thanks @bsongis @3djc ,problem solved.

closed 2019-07-30 by LOVEChen