opentx/opentx #6590: X-Lite conversion from 2.2.4N706 to 2.3.0N54 changed setup

notlebowski, 2019-07-28:
FYI

Yesterday I tested an upgrade from 2.2.4N706 to the 2.3.0N54.
Upon startup the X-Lite started to convert the firmware.
I then noticed that from at least my default model the setup has changed.

The flight modes were assigned totally different switches than on the original model.
I have not investigated other changes yet.

Please close this ticket if conversion errors should be expected at this stage of development.
I will close this ticket if there is no further interest.

I have an otx of before and after the conversion.
I can post those upon request.

[image]
[image]

3djc, 2019-07-28:
Yes please share your otx, this should not happen at this stage

notlebowski, 2019-07-28:
[OTX.zip](https://github.com/opentx/opentx/files/3438928/OTX.zip)

notlebowski, 2019-07-28:
During upgrading the X-Lite was close to unreadable.
I quickly took this picture in case it could be useful.

[image]

This afternoon I will downgrade my X-Lite, reload last week's setup and migrate again.

I forgot to mention I am also running Mike Blandford's Ersky9x's bootloader.
AFAIK this could have no bearing on this issue, but I'm not sure of that either.
If the outcome of the corrupted setup will be the same this afternoon I will also do a cold flash of the 2.2.4, load the model and upgrade to 2.3

3djc, 2019-07-28:
Reproduced (but not the display issue)

[image]

kilrah, 2019-07-28:
So is the conversion error only in companion or did you also have it on the radio?

The screen isue is normal, during conversion your custom contrast setting isn't available so a default is used, and that defaut will be either good of bad depending on the radio.

3djc, 2019-07-28:
Confirmed doing the conversion on radio

notlebowski, 2019-07-28:
I only did the conversion on the radio....

I wanted to reproduce it now, but now it has been confirmed by 3djc there's no need, I guess.
I'm assured you will find the culprit soon.

notlebowski, 2019-07-28:
My 2.2.4 setup was from last week, so I needed to reconstruct my 2.2.4 setup, so they now had the same subtrims and curves.
After that I did a "setup compare" and I noticed some more differences which may not be reflected in the new upgrade.

The RSSI-critical value (which I never touch) has suddenly changed from 33 to 42
The Top Bar settings were lost

[image]

The instances changed in sensors.
[image]

I named my SC-switch "Act" (for Action).
The "Ac" and "t" were then assigned to SB-switch and SC-switch
[image]

[image]

3djc, 2019-07-28:
The RSSI critical in your otx is set to 33 for 2.2.4

The Act seems to be a companion issue, will follow up, as the radio is ok

[image]

notlebowski, 2019-07-28:
Ah, ok.
I did not check that. It was terrible weather today for flying, but it was great for watching the GP F1 in Hockenheim.

PR #6591 (2019-07-30): Flighmode needs switch conversion too
This fixes #6590
closed 2019-07-30 by bsongis