opentx/opentx #6592: Companion 2.3 issues

3djc, 2019-07-28:
EEprom (xlite):

[eeprom-2000-01-30-085816.zip](https://github.com/opentx/opentx/files/3439423/eeprom-2000-01-30-085816.zip)

Issues : RSSI critical display 42 instead of 33
Hardware switch name for SC split in 2 (Ac and t)

lucifergeo, 2019-08-12:
This also applies for the X9-Lite.
both RSSI warning level are reset when importing/exporting Modells between companion and TX

notlebowski, 2019-08-13:
When I read models now, with radio upgraded to N61 I have a lot of empty lines in SF and probably much more is wrong

X-Lite classic

raphaelcoeffic, 2019-08-14:
@3djc with the changes in #6625, your eeprom works for me properly (tried with simu / companion).

raphaelcoeffic, 2019-08-14:
seems the special functions are still messed up:
[image]

raphaelcoeffic, 2019-08-14:
@3djc seems all the switches were affected. Xlite uses a number of switches in the storage different from what it has in reality (probably because of compatibility with X-lite S/Pro?).

3djc, 2019-08-14:
yes, most radio do that in 2.3, there is a NUM_SWITCHES and STORAGE_NUM_SWITCHES

closed 2019-08-14 by bsongis

notlebowski, 2019-08-16:
I just tested 2.3N62
The setup is still messed up if I read the radio with Companion.
The radio is still running N61 though.

Are these changes not yet in there or could it still be wrong?

kilrah, 2019-08-16:
N62 is from the 13th, this was merged in on the 14th.
Seems the build hasn't run.

PR #6625 (2019-08-14): Companion Fixes (#6592)
fixes #6592 

TODO:
- [x] fix offset in switch names.
- [x] load/save RSSI alarms
- [x] implement support for RSSI source.
- [x] fix special functions (see eeprom in #6592)
- [x] rename FrskyField to Frsky218Field and VariaFlield (without telem screens).