opentx/opentx #6594: X-lite Pro Gyro calibration and gyro(Stick) center beep

saladbutter, 2019-08-01:
Hi, i've got a paralysis on my right hand, that's the reason why i get a X-lite pro with 6 Axis gyro to usw ist on Mode 2 for Pitch&Roll.
My Idea: never give up. Love that Hobby.

On the Stick we all knew the Stick Center but not on the gyro, there isn't a Center beep. Only for Stick's.

Is it possible to get a Center beep here and is it possible to get a new (individual) gyro Center if i Switch from Stick to gyro Control?

It will be not funny if i arm the Quad and it goes crazy.

Thanks

kilrah, 2019-08-01:
You can make any beep you want with a logical switch e.g a~x GyrX 0 and a Beep special function.

Please note this is an issue tracker to report bugs and not for usage questions.

closed 2019-08-01 by kilrah

saladbutter, 2019-08-01:
Hi, das mit dem Center beep einstellen soll wie genau funktionieren ?
Und das mit den gyro Kalibrieren?

Wenn es die falsche Ecke war, wo muss ich das eröffnen?