opentx/opentx #6595: unify model backup format between radio types

lshems, 2019-08-02:
At the moment, an enormous amount of effort seems to be spent in getting companion to convert models from one radio type to another. With the rapidly increased number of radios out there, this becomes a very tricky and complex business.

But people rely on it for model exchange and other resource sharing stuff (yes, I use that as well for the distribution of my (free) models that can be set up using a (paid) LUA app).

Would it be possible to create an other mechanism to convert models between radio types?

Just to clarify:
if there would be access with LUA to all model settings, it would already be possible to create a backup file containing all model settings from one model using LUA, that could be loaded using LUA again on top of an empty model to recreate the first.
This would be completely radio independant, where the only thing to manage would be the remapping of switches, pots and trims, since different radios have different physical capabilities.

kilrah, 2019-08-02:
That's been in the plan for a long nwhile and may happen in 2.4. Pretty sure there are already open issues mentioning it.

It probably won't please you though since that will take more resources, and as a result will only be "general" for... the subset of existing and future radios with enough resources to run it without requiring compromises on existing stuff.

lshems, 2019-08-02:
Why would that not please me?

Sounds like a simplification still, no?

Eagerly waiting ..,

kilrah, 2019-08-02:
Because for quite a while it will not reduce the number of cases, it'll actually increase it by 1... 
Some radios that come out now still probably won't be able to get it.

raphaelcoeffic, 2019-08-04:
@kilrah if you are talking about the new storage, I think it can be done in a way that does not eat up so much ressources. The memory requirements are pretty low for now (and can probably be satisfied using the "reusable buffer"). The current YAML parser uses only very few memory. The biggest downsize is that a text based format would use much more of the eeprom storage, basically meaning that the SD card should be used instead (even on system that did not require it so far).
Anyway, even with that, due to the different hardware between the radios (different number of switches, etc), it is clear that some form of "conversions" will need to happen when loading a model that uses hardware not present on the destination transmitter.

lshems, 2019-08-04:
And that's why I linked this request to the generic layout request, which should cover that.

kilrah, 2019-08-04:
The generic layout by far won't solve everything. you just can't translate 8 switches into 4.

lshems, 2019-08-04:
Yes you can translate 8 switches into 4. You clearly didn't understand my proposal.

Create 8 switched, for which you may only have 4 physical inputs. If the not assigned switches would be able to be set to -100, 0 or 100 as a fixed value, no model transfer would go wrong. It's just that four of your switches are 'stuck' at the predefined values.

It's to the user to choose which model switches he wants to control with physical switches, and which not.

I know this is yet another layer, but the only way to simulate missing hardware switches.

You may also leave the non existing switch references and substitute them with a default value if evaluated. 

Several solutions possible to get rid of x*X*X configurations to manage.

kilrah, 2019-08-04:
> It's just that four of your switches are 'stuck' at the predefined values. It's to the user to choose which model switches he wants to control with physical switches, and which not.

Then it solves nothing, they'll still have to fix everything by hand.

> Several solutions possible to get rid of xXX configurations to manage.

For us not to have many configurations to manage is trivial and was figured out long ago, but it won't make it any more seamless or easier for the user.

lshems, 2019-08-05:
I disagree to all the above counter arguments.

It IS helpful to be able to load any model from any radio to any other model in any other radio using this scenario, instead of being forced to have a NON working model after conversion in companion where you need to check all switch setting or your model will crash.

But hey, opinions differ. I'm not in charge, you are.

kilrah, 2019-08-05:
It will STILL be non-working and requiring checking/changing all switches even with your method.

lshems, 2019-08-05:
It will be working, instead of having MISSING references. 

Just choose which switch functions you can miss in the lesser equiped radio. No expo switch? OK, set it on default -100? No Dual Rates? OK, set it on default -100. 2 missing flight modes? Ok, can live with that, set the switch to -100, 0 or 100, depending on the flightmode you like. No sound announcing altitude on the tempswitch? OK, can live with that, set it to -100.
Even if the missing switches are defaulted on loading to -100, it would just mean the user cannot change the state of some switches with a physical switch. But he can still change them.

I just reduced the number of needed physical switches by 4, without rendering the model useless. Even better, I can still set a default value for the switches I am missing, and can switch assignements if I would want to without modifying the model, releasing the fixed functions but fixing others. My choice.

I won't argue to get it implemented (or something in the same line of thinking), that's up to you, as I said.

But I don't agree with your assesment, and your arguments do not convince me at all.

I'll leave it to you to close the issue.

kilrah, 2019-08-05:
And how does it work when you go from a 3pos to a 2pos or vice versa?

lshems, 2019-08-05:
You'll simply have one position less. (I know that some edge applications won't work anymore, if you are fishing for that).

You must forget the entire physical two pos. Create a three pos slot for the max number of switches you think are useful, where you don't have a mid pos if a physical two pos switch is assigned. Could even assign a Gimbel to a switch slot.whats the problem?

As long as all slots are either assigned with a physical input or have a default value of -100, 0 or 100, all is fine.

?!

By the way, never understood why they would put a two pos switch where all is foreseen for three positions. I replace all two pos with a on-off-(on).

Gives all possible uses in one switch.

But that's off topic, true.

kilrah, 2019-08-06:
> (I know that some edge applications won't work anymore, if you are fishing for that).

Most commonly when you use a 3POS it's because you need the 3 positions. Your setup will be lacking something important when going to 2, so you'll have to reprogram your model to work with just 2.

If I look at my models it's not an "edge application", it's most of them.

mpjf01, 2019-08-06:
On the proposal to happen in 2.4 maybe that's been the plan for some time. Or what it morphs into. 

Right now when one wishes to use a model setup from anywhere not on his current transmitter he uses Companion to read the setup and Companion makes the changes that it can to accommodate any differences that there may be between the donor radio type and the user's radio. Anything that cannot be dealt with automatically is reported so that the user can make changes manually.

It's unclear to me what the procedure will be after the modifications mooted here. Could someone who understands it please explain what the user will have to do when he wishes to use a model setup created for a different transmitter type?

raphaelcoeffic, 2019-08-13:
@mpjf01 have a look at #6099, it has some examples of how the storage format will look like. In the meantime, let's close this issue. For those interested, the discussion shall continue in #6099.

closed 2019-08-13 by raphaelcoeffic