opentx/opentx #6596: Unify Switch Layout between radio types

lshems, 2019-08-02:
Supportive to the idea to make model exchange between different radio types easier, is the idea to make only ONE radio layout for switch, inputs, trims and pots.

The idea is that the hardware page would then list these generic entries, where you can then select to have them either controlled by one of the physically present input switches, or give them a default value.

In theory you could assign every physical input (switch, trim or gimbal input) to every general model standard input, as long as you map every physical input first to a range of -100 to 100.
You could then map a switch on the throttle input, a gimbal on a switch input, a pot as rudder input.
The hardware page would then create the logical layer between the controls of the actual radio and this general radio layout.

It would be defaulted to the layout as is currently 'hardwired', with the sole difference that for all physically not present switches, a standard value of -100 would already have been assigned for the general model switch.

models would be instantly exchangeable and useable between all radios without conversion logic.

If the idea is not clear, let me know and i'll try to clarify somewhat more.

If the idea is not interesting, or the problem of exchanging models between radio types is handled without the current drawbacks and manual checcking neccesary, please indicate that as well.

thanks for reading.

kilrah, 2019-08-02:
Already designed and planned for implementation.

closed 2019-08-02 by kilrah

lshems, 2019-08-02:
Doesn't mean it shouldn't be tracked and shared? No info that I can find here, so others probably neither.