opentx/opentx #6598: horus 12s and horus 10---8.3 filename in special functions

tj51, 2019-08-02:
if you go to System and find a mp3 with long filename, right click and play, it plays, go to model, special functions try to play track, go to the sound field and list the sounds, the long filename is missing out of the list, this function is in 8.3 mode not in ling file name mode, this is a problem, it works correctly on my tranis transmitter, but broken in Horus, I own a 12s, my friend owns a 10 and the same on both  horus's

3djc, 2019-08-03:
This is working as intended, the storage for filenames is limited, and is different on Horus because Horus allows several extensions

closed 2019-08-03 by 3djc

tj51, 2019-08-06:
in the system, sound the are about 25 characters in the file name, in the special functions the are only 8, this has nothing to do the the file extension to the right of the file name like .mp3 .wav, please expand the special function characters to match the system sound characters

kilrah, 2019-08-06:
System sounds are hardcoded in the firmware, so they can be any length without implication. 
User sounds must have their names stored in model memory.

3djc, 2019-08-06:
And also, for the sake of precision, it is not 8, it is 6