opentx/opentx #6599: Internal All Feature for modules on all radios

robthomson, 2019-08-03:
Currently opentx has an option called 'internalppm'.

This essentially feeds a ppm signal to the internal module bay.

We should open this to allow 'internal all' on all radios - not just radios like the xlite pro.

To give some rational - radios like X10S have an XJT module.  The module is not even solders in - but pluggable.

It would be very 'open' to allow modules to be swapped out.  So for example..  I love my X10S.    However.. I fly lots of modules that support crossfire.. in the rear bay.  But dont really want to constantly swap modules so my DSMX kit will work.

So... along comes an internal multiprotocol module.

Plug it in - and magically my radio does multi and crossfire.  or multi and r9 etc..

This option could be achieved as a companion 'compile time' option.  So we have an 'internal ppm'  and we have an 'internal all'

etc..

Make sense?

kilrah, 2019-08-03:
It's possible for sure, but we never went more than PPM because of the UI nightmare that would be.
But we're slowly making that less of a convoluted mess, so maybe...

robthomson, 2019-08-03:
Maybe just drop internal ppm.  And leave 'internal all' as an option?

kilrah, 2019-08-03:
The whole point is that making "all" protocols available on internal is a lot of work and pain, when almost nobody will use it...

robthomson, 2019-08-03:
This is true.. but I would hazard a guess that 99% of the work is already done and could be pulled from a different git tree :-)

3djc, 2019-08-03:
but that been 2.2 based, is not really compatible with 2.3

robthomson, 2019-08-03:
Ah.,. I was not aware 2.3 changes so much!

kilrah, 2019-08-03:
No, we never wrote pulse drivers for anything other than PPM or PXX on the peripherals for the internal module. So 4 drivers to write, several times becasue not all radios use the same peripherals. The the UI, which also has to take into account the incompatible internal/external combinations,...

We're not against it at all, if you want to code it we're totally happy to include it :)

robthomson, 2019-10-02:
Ok.. so it seems.. without being too researched that:

https://www.jumper.xyz/portal.php?mod=view&aid=17

Has a 2.3 firmware that includes the new internal module features.

Maybe  time to drag this up again?

kilrah, 2019-10-02:
It's coming in 2.3.

closed 2019-10-02 by kilrah