opentx/opentx #6600: cannot bind with telemetry off on X9 Lite

P-I-Engineer, 2019-08-03:
**Describe the bug**
bind with telemetry off still keeps telemetry on

**To Reproduce**
Steps to reproduce the behavior:
bind
telem off
power cycle rx

**Expected behavior**
radio should not report rssi

**General information**
 - OpenTX version 2.3 N57
 - X9 lite
 - tried on XSR and R-XSR
-confirmed rx's work correctly by testing on my taranis X9D

Lasakro, 2019-09-22:
Any progress on this? I'm having the same problem with two X8R's and one RX8R with the latest firmware running OpenTx 2.2.4 on a X9D+.
X8R FW- 2-10-2017
RX8R FW- 6-1-2018
Should this problem be brought up with FrSky, maybe receiver firmware problem?

It really causes problems when running a redundant receiver and you are looking for consistent telemetry.

P-I-Engineer, 2019-09-22:
I haven't tested on my xsr yet, but, I upgraded the internal frsky module firmware and it fixed my other issue with not being able to bind with telem on and channels 9-16 on my rx6r.

So, update all your firmwares

Lasakro, 2019-09-22:
Updated XJT Internal RF module, problem solved. Thanks!

closed 2019-09-22 by kilrah