opentx/opentx #6603: Custom telemetry screens: XLite Pro & Companion

MikeNH, 2019-08-06:
**Describe the bug**
Taranis Xlite Pro and cannot get more than 1 custom telemetry screen to present itself. This the same behavior for XLite Pro and OpenTX Companion Simulator. 

I've seen this behavior for many nightly releases and am surprised that I haven''t seen this on RC Groups or as an open issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Using Companion (but also is seen on the radio), create a new multicopter model and accept all defaults.
2. Open new model and go to Telemetry tab.
3. Scroll down to the bottom and on Telemetry screen 1, for Custom Screen Type drop-down, choose Numbers. From dropdown, choose Batt.
4. Telemetry screen 2, for Custom Screen Type drop-down, choose Bars. From dropdown, choose Batt.
5. Click Simulate
6. On Simulator, long-press the page down button to display 1st custom screen. 
7. On Simulator, hold down the "s" (shift) key and at the same time, click the left or right page button.
At various times the screen will flicker both on the radio and on Companion indicating an attempt to display screen 2 but the next configured never displays.

**Expected behavior**
Custom telemetry screens 2,3 and 4 should display if configured.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version2.3.0N58
 - Hardware XLite Pro
 - OpenTX Companion Windows 10

kilrah, 2019-08-06:
Wrong issue description, it's not that you can't change screen, it's that the "Bars" screens are broken. If your 2nd screen is Nums it switches fine, and if you select only one screen that is Bars you et "no telemetry screens".

kilrah, 2019-08-06:
Scratch that, it's all working correctly. Looks like you forgot to set the min and max values for the bar.

closed 2019-08-06 by kilrah

MikeNH, 2019-08-06:
Sorry but the problem exists as an example when screen 1 & 2 are set for numbers and both use the "thr" parameter. Only the first screen appears and correctly reflects the throttle setting. No min/max settings required.

kilrah, 2019-08-06:
Works just fine here...