opentx/opentx #6605: Companion 2.3.0N58 - Open .otx from 2.2.4 Horus X10s to convert in a new in X-Lite S Profie chrash

SnappyFPV, 2019-08-06:
I did save this Model Config .otx in 2.2.4 from my X10S to convert a model to my X-Lite S in 2.3.0N58  after the Message Box confirm the Companion App chrash and close.

[Test.zip](https://github.com/opentx/opentx/files/3474001/Test.zip)
[image]

 - OpenTX Companion 2.3.0N58

3djc, 2019-08-06:
While xlite are supported, Horus aren’t currently supported in 2.3

closed 2019-08-06 by 3djc

SnappyFPV, 2019-08-06:
So no chance to convert model at the moment?

kilrah, 2019-08-06:
No. There is nothing you could do with it anyway since Horus 2.3 is far from being usable in the first place and will just crash on you.

SnappyFPV, 2019-08-06:
Thanks - will do it from hand

3djc, 2019-08-06:
Only between currently supported models

kilrah, 2019-08-06:
What you do from hand is not guaranteed to work once we start provifding builds, and again **you should not use 2.3 for Horus for anything at this point**.