opentx/opentx #6606: backlight not working with x-lite s

nitram509, 2019-08-08:
**Describe the bug**
Newly flashed firmware 2.2.4 on X-Lite S does not light on the backlight anymore.
Backlight remains off.
No settings have any influence (brightness level set to 100 has no effect).

**To Reproduce**
Steps to reproduce the behavior:
1. Just switch on the radio - screen remains off / no light
2. changing settings in brightness or automatic switch off has no effect at all.

**Expected behavior**
backlight should be on and can be controlled via brightness setting.

**Screenshots**
n/a

**General information**
 - OpenTX version 2.2.4, opentx-xlite-noheli-lua-luac-sqlfont-en
 - Hardware X-Lite S
 - OpenTX Companion 2.2.4, OSX, 64bit

kilrah, 2019-08-08:
The X-Lite S is a new radio with different hardware and it is not supported in OpenTX 2.2. Need the X-Lite S radio type in OpenTX 2.3.

closed 2019-08-08 by kilrah

Lenio-frihling, 2019-08-17:
I have the same problem with my X-lite S. How to return to opentx 2.3?

nitram509, 2019-08-18:
@Lenio-frihling the OpenTX 2.3 is still under development and AFAIK no preview release available.
You need to download the source code an compile on your own ... e.g. following these instructions ... https://github.com/opentx/opentx/wiki/OpenTX-2.3-Docker-Build-Instructions

3djc, 2019-08-18:
What are you talking about ? 2.3 nightly are available, there is absolutely no need for self compilation...

nitram509, 2019-08-18:
Thank you for clarification. I have overseen them.
here's the link http://www.open-tx.org/2019/05/17/2.3-nightlies

kilrah, 2019-08-18:
He also could have just relaoded the stock firmware from FrSky's website...

404Error404, 2019-08-22:
hi, i just flash 2.2.4 on my xliteS and so i understand it was an arror. 
where can we find the original firmware of xlite S pls ? 
when i download opentx 2.2.3 it doesn't download 2.2.3 firmware and then no xlite S option... im lost

nitram509, 2019-08-23:
@404Error404 the original firmware you need to download from FrSky's website.
If you want to run OpenTX, please download "2.3" instead of "2.2.3" - because the 2.2.x releases don't support the Xlite S/Pro version.

404Error404, 2019-08-23:
thanks for response,
my mistake is to use 2.3 nighlty for download. 
the right way is to use "transfer" button on the left of companion and then select FrSky's website files for xlite S... 
my radio work fine again, thanks.

Luckyluke1988, 2021-11-09:
Hi, i just updated my Taranis FrSky X Lite Pro to version 2.3.14-otx (1e09791a) downloaded from Companion 2.3.14 and now the Backlight is off, the Settings it is turned on. Any ideas?

Lenio-frihling, 2021-11-10:
> Oi, acabei de atualizar meu Taranis FrSky X Lite Pro para a versão 2.3.14-otx ( [1e09791](https://github.com/opentx/opentx/commit/1e09791a1e2fe2a0ca9835019d634a4c6a4fa3bf) ) baixado do Companion 2.3.14 e agora a luz de fundo está desligada, as configurações estão ativadas. Alguma ideia?

When I had this problem it was because I installed the firmware for X-lite (simple) on X-lite S.
Pay attention, make sure you are using the correct firmware.