opentx/opentx #6610: New Taranis X9D PLUS SE 2019 on 2.3 no crossfire support

craigywsm, 2019-08-10:
Hi All,
Just purchased the new 2019 Taranis and in the external module there is no option for crossfire. please can someone help me support this.
https://www.youtube.com/watch?v=WhZTXuKV4TI&t=1s

3djc, 2019-08-10:
We are currently not supporting X9D 2019. We understand FrSky provides a crippled down version of OpenTX, but on our side, we will not start supportng a radio until it passes our internal testing, and not before all the features our users expect (Crossfire, multi, companion,..) have also been finished and tested.

FrSky did not want to supply samples for the X9-Lite to any member of the team other than their employee, and even worse that they did not allow them to see the technical docs before public release. That has reduced our ability to provide on time support for this radio, and will likely complicate ongoing support as well

closed 2019-08-10 by 3djc

craigywsm, 2019-08-10:
3djc thank you for your information.
Looks like FrSky are alienating themselves on this one.
Looks like they could be loosing alot of customers. 
I must admit i am coming from another hardware manufacture but like the Open TX firmware

3djc, 2019-08-10:
We love crossfire too ! As soon as this radio will be supported by us, crossfire will be available :)

craigywsm, 2019-08-15:
Hi there,
thanks for the fast write to support the new 2019 edition with the latest
nightly build.
FYI good news looks like crossfire works.
I have a YouTube video if you want to have a look?
Hope for a nice release candidate of 2.3 soon.
Thanks for all your hard work in the community.

Kind Regards,

*Craig Reynolds*

IT Consultant

M:   +44 (0)7725 216 002
E:    craigywsm@gmail.com

[image: http://uk.linkedin.com/pub/craig-reynolds/70/215/792/]
<http://uk.linkedin.com/pub/craig-reynolds/70/215/792/>

This e-mail contains confidential information and is for the exclusive use
of the addressee/s. If you are not the addressee, then any distribution,
copying or use of this e-mail is prohibited. If you recieve in error,
please advise the sender and delete it immediately. We accept no liability
for any loss or damage suffered by any person arising from use of this
e-mail.

 P Think of the environment; please don't print this e-mail unless you
really need to.

3djc, 2019-08-15:
We need to work on Horus before we can progres toward RC, but that will come

craigywsm, 2019-09-04:
Hi,
just keeping up with things and noticed that crossfire option has
disappeared from the latest nightly companion.
Please can you let me know what the issue is?

Kind Regards,

*Craig Reynolds*

IT Consultant

M:   +44 (0)7725 216 002
E:    craigywsm@gmail.com

[image: http://uk.linkedin.com/pub/craig-reynolds/70/215/792/]
<http://uk.linkedin.com/pub/craig-reynolds/70/215/792/>

This e-mail contains confidential information and is for the exclusive use
of the addressee/s. If you are not the addressee, then any distribution,
copying or use of this e-mail is prohibited. If you recieve in error,
please advise the sender and delete it immediately. We accept no liability
for any loss or damage suffered by any person arising from use of this
e-mail.

 P Think of the environment; please don't print this e-mail unless you
really need to.

kilrah, 2019-09-04:
None, it's just always included now.

craigywsm, 2019-09-05:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/opentx/opentx/issues/6610?email_source=notifications\u0026email_token=AJTZRQY3727A6GAZXXCOW7DQIA45XA5CNFSM4IKZF5WKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD55IXYQ#issuecomment-528124898",
> "url": "https://github.com/opentx/opentx/issues/6610?email_source=notifications\u0026email_token=AJTZRQY3727A6GAZXXCOW7DQIA45XA5CNFSM4IKZF5WKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD55IXYQ#issuecomment-528124898",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

I have just updated and the crossfire tick box is missing. Will it come back?Kind RegardsCraigSent from my Samsung Galaxy smartphone.

3djc, 2019-09-05:
Please read my comment just above. It will NOT come back, because Crossfire and multi are now part of every firmware, no need to tick it anymore

craigywsm, 2019-09-05:
Great thanksKind RegardsCraigSent from my Samsung Galaxy smartphone.

[
{

Noidea247, 2019-09-10:
This is my first post here so be gentle but I also have the latest x9d 2019 taranis (same as yours Craig I’ve watched your video) but for the life of me I cannot get crossfire to work. I have the latest 2.3 companion update (n82 I think) but whenever I try and flash new firmware it says my hardware isn’t compatible even though I have the correct radio selected. The radio is on and in “sd mode” and I can read the stock firmware fine. But as soon as I try and flash the new one, nothing. Any ideas? Thanks.

kilrah, 2019-09-10:
Did you correctly select the new "X9D+ 2019" type in companion insetad of the default "X9D+"?

craigywsm, 2019-09-10:
Please check what kilrah said and let us know the outcome.

Noidea247, 2019-09-11:
Yes the 2019 version of the radio is selected. It’s exactly as follows in the radio type “frsky taranis X9D+ 2019”

Noidea247, 2019-09-11:
The exact error when I try and write new firmware (which is “OpenTX-x9d+2019-lua-noheli-en.bin” which as of today is version 2.3.0N84) says “New firmware is not compatible with the one currently installed”

3djc, 2019-09-11:
Try unchecking "Check hardware Compatibility"

Noidea247, 2019-09-11:
Done, error “Cannot write E://Firmware.Bin” (reason: Unknown error)

3djc, 2019-09-11:
Then you are not corretly following the upgrate procedure. What  exactly do you mean by " The radio is on and in “sd mode”"

Noidea247, 2019-09-11:
I’m assuming I am supposed to be doing this with the radio turned on and in “sd card” mode. Just as I am when I read the firmware? Companion can read it perfectly fine FYI. I have made several “test” backups without issue.

3djc, 2019-09-11:
what do you call 'sd card' mode ?

Noidea247, 2019-09-11:
Ie I can see the screen with a small usb symbol in the corner. From what I’ve read this radio doesn’t need to be turned on in any special way like other taranis’s. From what I’ve read of course.

Noidea247, 2019-09-11:
When I plug the USB into the radio while it’s turned on it asks me do I want to go into simulator mode or sd mode.

Noidea247, 2019-09-11:
Ok so I figured it out sorry gents. 
I did need to turn the radio on in boot loader (different from what I’ve read but obviously my mistake) 
I still needed to uncheck “check hardware “ but the firmware did flash. 
I know the nightly builds are unstable and my radio now says nightly build do not fly when I turn it on but it does actually work right?

3djc, 2019-09-11:
OK, thats not the right mode for flashing, you need to be in bootloader mode, achieved by starting the radio with both horizontal mode pinched toward center, as any other Taranis. There are plenty of tutorial video and sites explaining that