opentx/opentx #6612: Invalid Logfile: X9Lite ( 2.3.0N61)

r41065, 2019-08-11:
Trying to open logfiles with Companion internal LogView 
i get always an Error message:
" The selected Logfile contains xxxx invalid lines out of xxxxtotal lines "

[image]

openingthe .csv file for example with libre office it looks OK at the first view.

opening with a textbrowser i found out that the headline doesn't match to the number of Colums:
extract from Headline:
RxBt(V),Rud,Ele,Thr,Ail,S1,SA,SB**,SC,SE,SF**,LSW,TxBat(V)
extract from example Line:
4.8,-3,-1,-1024,1,-2,0,-1,**-1,-1**,0xFFFFFFFF00000000,7.2

in Headline is missing switch SD:
 there are listed 3 switches SC, SE, SF:

in example line only listed the values of 2 switches ( i didn't find out which ones till now...) at same Position, Values from SA, SB and LSW seemed to be OK

The X9Lite doesnt have switch SF, so if i delete the text " SF, " in headline, will be able to open the Logfile without warning...

Ralf

ps. cant open Logfiles from X9Lite since the Frsky Version my TX was delivered with...

r41065, 2019-08-11:
complete corrupted logfile:
[1RK_ACCESS-2019-08-11.zip](https://github.com/opentx/opentx/files/3490278/1RK_ACCESS-2019-08-11.zip)

3djc, 2019-08-11:
The fix for this is ready in PR #6611.

PR #6611 (2019-08-12): Logs : log all switches based on hardware config
- Display either user defined switch name if exist or predefined siwtch name in column header
- Logs all switches available based on hardwrae configuration
- No more (ugly) per radio predefinition
- This also fixes the issue where X9Lite logs cannot be read by Companion
- This fixes #6612 

Sample from X9E testing

`Date,Time,Rud,Ele,Thr,Ail,F1,F2,F3,F4,S1,S2,LS,RS,SA,SB,SC,SD,SE,SF,SG,SH,SO,SP,aaa,LSW,TxBat(V)
2019-08-10,17:28:16.960,0,0,-1024,0,0,0,0,0,0,0,0,0,-1,-1,-1,-1,-1,-1,-1,1,1,1,0,0x0000000000000000,10.6
2019-08-10,17:28:17.820,0,0,-10 [... cut]
closed 2019-08-12 by bsongis