opentx/opentx #6613: X9 Lite - R9M Lite Pro | No indication of Rx connection

ijustwannafly, 2019-08-11:
**Describe the bug**
Usually on my xsr series receivers, I will have an RSSI "bar" display to the immediate right of the battery voltage on the main screen.  On disconnect/reconnect of rx link "telemetry lost/telemetry recovered" is audible/initiated properly.  

With the X9 Lite using the R9M Lite Pro flashed to FLEX ACCESS 1.1.1 there are no signal bars and no "telemetry lost/telemetry recovered" audio events under the proper circumstances.

**To Reproduce**
Steps to reproduce the behavior:
1. X9 Lite currently on 2.3N57 - R9M Lite Pro FLEX ACCESS 1.1.1
2. Establish link (Reg, bind, assign channels/functions)
3. Verify link works properly - I verified this by arming and hovering one of my quads.
4. Disconnect battery, no telemetry lost event.  Reconnect battery, no telemetry recovered event.

**Expected behavior**
Signal bars for simple RSSI on the main screen when rx link is established (as it behaves with ACCST xsr series receivers) and telemetry lost/recovered events.

**General information**
 - OpenTX version 2.3N57
 - X9 Lite
 - R9M Lite Pro FLEX ACCESS 1.1.1
-  R9 Mini flashed with FLEX ACCESS firmware

kilrah, 2019-08-12:
We cannot do anything about issues with FrSky modules/receiver misbehavings. If the module doesn't give us an RSSI we can't display it. 
It's been reported that ACCESS Flex firmware has issues, you can try relaying this to FrSky.

kilrah, 2019-08-12:
Correction, we simply don't support eu/flex for access at this time.

kilrah, 2019-08-12:
Works fine with Lite, someone with access to one will  have to test on Lite Pro. There might be an issue with the module since both should appear the same from our position.

ijustwannafly, 2019-08-12:
Yes, this behavior does work on the r9m lite with ACCESS FLEX firmware on my x9 lite with otx 2.3 nightly.

I have both the r9m lite and the r9m lite pro and the pro is not producing the same behavior.

kilrah, 2019-08-13:
OK good to know, thanks. Unfortunately none of us has a Pro module at this point, so it'll likely be a while.

ijustwannafly, 2019-08-13:
If I can help anywhere let me know.  The x9 lite is currently my primary radio since I actually prefer it to my x-lite (shocker) so I went through the effort of flashing my rx's to ACCESS firmwares where supported.  The r9 setups are on FLEX because it was working that way for me on FLEX ACCST.

ijustwannafly, 2019-08-15:
Something interesting.  I reflashed the same receiver with the same firmware for FLEX ACCESS.  It still showed up as registered to the lite pro module so I did a reset on the receiver from the tx.  Then I re-registered the rx to the module and did a re-bind.

Now it behaves properly.  

I will close this issue and apologize for my lack of proper testing prior to opening this issue.  :)

closed 2019-08-15 by ijustwannafly

3djc, 2019-08-15:
Not the first time we see or hear about similar situation with ACCESS. There seems to be something on FrSky side that is sometime behaving wrong