opentx/opentx #6614: Qx7 smart port mirror

KnuckleUpFPV, 2019-08-11:
Is it possible to get a crsf mirror and smart port mirror on the qx7 like the x9d has? I know others have mentioned on the x9d smart port mirror about using micro tx crossfire for their trackers. The qx7 doesnt have the option displayed for the telemetry mirror to use it with bluetooth. If this isn't able to be done please let me know, and I will delete this.

pulquero, 2020-02-12:
Have you tried using the smart port on the bottom of the q x7?

noemu, 2020-03-07:
Without changing any settings?

pulquero, 2020-03-07:
Yes.

noemu, 2020-03-11:
someone has a clue, how the packets are structured?

(i'm using inverted tx/rx  57600, N, 8, 1. )

example data
` 20 09 00 a0 51 00 a0 42 44 80 84 05 84 52 00 92    ...Q..BD....R..
 00 20 08 00 a0 42 44 80 84 41 81 a4 c0 02 00 a0   . ...BD..A......
 42 44 88 40 40 80 80 00 45 00 a0 48 00 20 52 00   BD.@@...E..H. R.
 a0 42 44 a0 84 88 81 00 00 29 00 a0 0a 00 a0 42   .BD......).....B
 44 a8 82 00 00 00 00 89 00 a0 42 44 80 40 40 80   D.........BD.@@.
 80 00 25 00 20 25 00 20 22 44 81 04 80 81 11 00   ..%. %. "D......
 22 00 20 22 44 41 04 80 80 00 00 18 00 20 22 44   ". "DA....... "D
 40 04 90 80 00 00 84 00 20 22 44 82 04 a2 80 00   @....... "D.....
 00 a5 00 a0 42 44 88 40 82 80 00 00 54 00 a0 42   ....BD.@....T..B
 44 90 40 41 00 00 00 4a 00 a0 42 44 80 82 02 00   D.@A...J..BD....
 00 00 29 00 a0 42 44 88 82 00 00 00 00 49 00 a0   ..)..BD......I..`

kilrah, 2020-03-11:
What is it supposed to be?
You can see in OpenTX code how the various protocols are parsed.

noemu, 2020-03-13:
today i tried to read the data again and it finally looks like s.port data. (maybe i connected it wrong before)

### More Info:
* the pins are from left to right:  GND V3.3 Duplex
* _Hardware:_ http://www.robotmaker.eu/ROBOTmaker/quadcopter-3d-proximity-sensing/frysky-s-port-interface (to set the Rx/Tx Signal as inverted signals, you need MProg from ftdi)
* _How to decrypt:_ https://github.com/opentx/opentx/blob/42a59dca3515f6e55a97ab422860b73f2194cbb6/radio/util/sport-parse.py
* _Sensor Id's:_ https://github.com/opentx/opentx/blob/42a59dca3515f6e55a97ab422860b73f2194cbb6/radio/src/telemetry/frsky.h

### example Data
```
7e 1b 10 10 02 2a 00 00 00 b3 
7e 98 10 05 f1 00 00 00 00 f8 
7e 98 10 01 f1 69 00 00 00 93 
...
```

`7e` start_stop byte
`1b` data_id byte
`10` frame_byte
`10 02` Data_Id (in little endian: VFAS_FIRST_ID <= `0x0210`  <= VFAS_LAST_ID )   
`2a 00 00 00` : data = 45 (it's 0.45 Volt (connected via usb))
 `b3` some crc checksum

3djc, 2020-03-13:
Closing this as physical pin and BT are both available on X7

closed 2020-03-13 by 3djc