opentx/opentx #6624: [call for review] updated wiki, docker build instructions

nitram509, 2019-08-13:
Hi,

I did create a wiki page on how to compile the firmware using the docker image.
This is a great way for many developers since it works without preparing all the required tools and results in reliable builds.

Because I'm not a native English speaker, I would like to kindly ask for review/feedback of this page:
https://github.com/opentx/opentx/wiki/OpenTX-Docker-Build-Instructions

Regards
Martin

lshems, 2019-08-13:
I, as many others, cannot use docker as it is not free?

raphaelcoeffic, 2019-08-13:
> I, as many others, cannot use docker as it is not free?

@Ishems why do you think it's not free? I'm using it every day, and it's open source with project Moby.

raphaelcoeffic, 2019-08-13:
@nitram509 I haven't tried it out myself, but you might use the same mount path for both commands, as some might think it should different when executing `cmake` and then with `make`. It could be a good idea as well to add some instructions for windows (mount path is different).

elecpower, 2019-08-13:
@nitram509 my suggestions

- avoid abbreviations e.g. _configuration_ instead of _config_ to help in text translations
- add the applicable Open Tx version(s) to the page title as the instructions can change between versions
- I know this should not be the case but based on previous misunderstandings, the instructions need to be clearly sequential and for multiple operating systems
- make prerequisites its own section
- add a link to the main Docker site so responsibility for installing and current legal use is up to the individual

Introduction

- _This little tutorial_ suggest _These instructions_
- typo _radion_ should be _radio_

**build docker image**

- change _tooling_ to _tool chain_ for consistency with other parts of the Wiki
- add a link to the how to create a local clone of the Open Tx repo (move to prerequisites?)

**Build the firmware**

- as @raphaelcoeffic 
- how to transfer the newly compiled firmware onto the radio. You would be surprised how many do not know how to do this.

Thank you for the effort

nitram509, 2019-08-15:
Thank you all for your feedback. 👍 
I did update the page accordingly.

If there's nothing left to say ... thank you and please close this issue :)

lshems, 2019-08-15:
About the opensource aspect of docker.

I was deviated to the paid version, as happens with so many opensource projects, had to register, only to find out only for companies?

I'll try and find the opensource version.

raphaelcoeffic, 2019-08-15:
It's called "community edition" ("ce").

raphaelcoeffic, 2019-08-24:
@nitram509 I guess we’re done here? Let’s close then.

nitram509, 2019-08-25:
@raphaelcoeffic can be closed, yes.
Thank you all for you support.

closed 2019-08-25 by nitram509

raphaelcoeffic, 2019-08-31:
@nitram509 the page looks like it has disappeared. What happened?

kilrah, 2019-08-31:
For some reason he renamed it, so it broke existing links... https://github.com/opentx/opentx/wiki/OpenTX-2.3-Docker-Build-Instructions

nitram509, 2019-08-31:
@raphaelcoeffic I renamed the page, because of feedback (see above)
"add the applicable Open Tx version(s) to the page title as the instructions can change between versions".

kilrah, 2019-08-31:
OK, the index page wasn't updated for it though. Just did that.