opentx/opentx #6631: X9D 2.3N062 wrong LS conversion

Allerhopp, 2019-08-15:
Radio converts telemetry value (Distance) in logic switches (V1) into Input 15.
Companion converts correctly.

[image]

raphaelcoeffic, 2019-08-15:
What do you mean by “convert”? Was it an existing model you converted from 2.2.x firmware? It would be nice to be more explicit with the steps necessary to reproduce.

Allerhopp, 2019-08-15:
Yes

raphaelcoeffic, 2019-08-21:
@Allerhopp I've tested manually (+added automated test) and it seems that the issue is not with the radio, where it just works fine, but with companion itself. Once the model is loaded into companion, the source is not displayed.

This is what I entered in Companion 2.2.4:
[image]

And here is how it looks like with current 2.3 code:
[image]

On the simu with 2.3 X9D+, it looks just fine:
[image]

notlebowski, 2019-08-21:
I am using delta as well to give me an altitude callout each 20 metres and noticed that wasn't working anymore. I reconstructed that part of my setup.

notlebowski, 2019-08-21:
Is it possible that this is already fixed in 2.3N67?
I just read a model which uses a delta together with Altitude and it reads this correctly.

raphaelcoeffic, 2019-08-21:
@notlebowski nope, if you open a 2.2 model in 2.3 companion, the bug is still there. Should be fixed shortly though, I'm on it.

notlebowski, 2019-08-22:
Both radio and companion are on 2.3 in my setup nowadays.
I am not sure when this conversion error happened.
Maybe I read a setup once after my radio was brought back to 2.2 because I had other conversion errors.
I can't read 2.2 otx files with Companion 2.3 at this moment.
This means I can't use the "compare tool" to verify my models.

At the moment I'm only reading setups in Companion from my radio and do all the editing on the radio as I'm afraid to introduce some errors due to Companion.

This doesn't mean Companion is doing anything wrong now (I'm running nightly N67 and don't build from source anymore).

PR #6652 (2019-08-23): [WiP] companion storage tests
More tests needed to be useful.

closes #6631
closed 2019-08-23 by bsongis