opentx/opentx #6633: X9 Lite N62, multiprotol says INV. PROTOCOL

P-I-Engineer, 2019-08-16:
**Describe the bug**
multiprotocol not working. 
all protocols say INV. PROTOCOL

**To Reproduce**
got to ext module
change to multi

changed back to N57 and it works.

**General information**
 - OpenTX version 2.3N62
 - Hardware X9 lite

Talha909, 2019-08-17:
I have the same issue. I have xlite Pro. I have installed the Open TX N62. In N62 version my multi module works good and come online but when I updated the Open Tx N63. My Multi Module goes offline and and in setup screen it gave me the fallowing status  Prot. invalid.

raphaelcoeffic, 2019-08-17:
```((fc0e7294a...)|BISECTING) $ git bisect good
fad8dd19269388146390adb2aec93d655f653157 is the first bad commit
commit fad8dd19269388146390adb2aec93d655f653157
Author: Bertrand Songis <bsongis@gmail.com>
Date:   Wed Aug 14 12:34:44 2019 +0200

    [X10 Express] Fixes
    - Introduction of -DPCBREV=EXPRESS
    - X10 Express external pulses all good
    Caution: need to check if no regression on X10 STD
```

raphaelcoeffic, 2019-08-17:
9294ec195fe629eff0e63a235c8928dd9e81c6d5 fixes it for me

raphaelcoeffic, 2019-08-17:
@bsongis would you mind having a look?

closed 2019-08-18 by bsongis

Talha909, 2019-08-18:
HI 
I have installed the N64 and the result was same. There is no communication with Multi Module .

bsongis, 2019-08-18:
It has been pushed this morning, it will be in next nightly builds

commit f1890bde (2019-08-18): Multiprotocol broken - Fixes #6633