opentx/opentx #6636: Error downloading firmware 2.3 for x-lite s

Lenio-frihling, 2019-08-18:
When attempting to download firmware 2.3 by companion the following error is reported: "No firmware release builds are currently being served for this version, lease switch release channel".

3djc, 2019-08-18:
You need to use nighlty channel

[image]

closed 2019-08-18 by 3djc

404Error404, 2019-08-23:
pls where is the original xlite S firmware... i delete it and flash 2.2.4... big mistake... now i lost backlight and instant switch.... pls where is guenuine xlite S firmware....why it's so hard to find it ?

3djc, 2019-08-23:
Is it ? For year, OpenTX firmware have been  downloaded though companion, this is no change. Download companion 2.3, setup your radio type and your preferences, and download...

kilrah, 2019-08-23:
And the one that was on the radio when you received it is in the downloads section for that radio on FrSky's website as you'd expect, what's hard about that?

404Error404, 2019-08-23:
im an idiot...
im looking for eeprom and firmware files where it's a bin and dfu files... 
another mistake is download with compagnon that is not the right files for S version or it's nighlty one.
the right way is to use "transfer firmware" button on the left of compagnion and then select FrSky's website files for xlite S. 
my radio working again. thanks.